chuot/rdio-scanner #1: website access to dev instance from another computer

swyman18, 2019-07-12:
Hi there, I was able to get a development instance running and it seems to work if I access http://localhost:3000 from that same computer.  I was hoping to be able to access that instance from another PC on my network using http://192.168.#.#:3000.  When I try to access the webpage using the ip address, I just get a blank page with "Rdio Scanner" showing in the browser tab.  Is there a setting somewhere that will allow me to access the webpage from another PC on my network?

Thanks!

chuot, 2019-07-12:
Hi,

Running the Rdio Scanner in development mode is intended to be used locally only with an IDE such as Visual Studio Code. However, you will have to tell Meteor what ROOT_URL to use instead of the default `localhost:3000`. You can try `ROOL_URL=http://192.168.x.x:3000/ npm start`, but only for test purposes and never for the production mode. In development mode, the code is compiled for debuging and neither optimized nor secure.

My recommendation is to build the application with the command `meteor build ../build` and install this new *compiled* package on your server. It will still run on port 3000 but optimized for production and secure. That is also fine if you only planned to access it only from your local network with less than 3 clients. Ideally, you should have a reverse proxy that, which would serves Rdio Scanner to the outside.

Note that Rdio Scanner settings (not meteor ones) is done via the `private/settings.json` file, when it runs in development mode, but is done with environment variables in staging/production mode. Check out the ̀`examples/rdio-scanner.service` file, you'll see some environment variables that tells Meteor how to behave with IP/port and API keys for uploads.

Keep us informed of your progress.

swyman18, 2019-07-12:
The ROOT_URL did the trick, thanks!  That will suffice for my initial testing for now.

closed 2019-07-12 by swyman18