chuot/rdio-scanner #2: Can't enforce local install with npm

chazanov, 2019-08-18:
I'm should not getting these errors from my npm 6.10.3:
```
npm ERR! 404 Not Found - GET https://registry.npmjs.org/rdio-scanner - Not found
npm ERR! 404  'rdio-scanner@1.0.0' is not in the npm registry.
```
How to avoid this?

chazanov, 2019-08-23:
I would like to package rdio-scanner for ArchLinux. So I'm doing:

`npm install -g --prefix "$pkgdir/usr"`

but it fails:

`npm ERR! Can't install ./rdio-scanner: Missing package name`

chuot, 2019-08-24:
After you have cloned this repository and installed the prerequisites, you must create an installable package from the source. This is done with the `meteor build ../ build` command according to README.md instructions. Then you can extract `../ build / rdio-scanner.tar.gz` where you want and start Rdio Scanner from it.

closed 2019-08-29 by chuot