chuot/rdio-scanner #3: Attempted install - failure building client app

kswanson71, 2019-11-27:
Installing on Ubuntu 18.04 Server and followed install instructions... am I missing something?

Snippet from failed install..

[image]

Thoughts?

chuot, 2019-11-27:
Go inside the client folder and try: npm run build

The resulting log should give us a clue.

kswanson71, 2019-11-27:
radio@rdio:~$ cd rdio-scanner/
radio@rdio:~/rdio-scanner$ npm run build
npm ERR! Linux 4.15.0-70-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2

npm ERR! missing script: build
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     /home/radio/rdio-scanner/npm-debug.log
radio@rdio:~/rdio-scanner$ more npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'build' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose stack Error: missing script: build
4 verbose stack     at run (/usr/share/npm/lib/run-script.js:147:19)
4 verbose stack     at /usr/share/npm/lib/run-script.js:57:5
4 verbose stack     at /usr/share/npm/node_modules/read-package-json/read-json.j
s:345:5
4 verbose stack     at checkBinReferences_ (/usr/share/npm/node_modules/read-pac
kage-json/read-json.js:309:45)
4 verbose stack     at final (/usr/share/npm/node_modules/read-package-json/read
-json.js:343:3)
4 verbose stack     at then (/usr/share/npm/node_modules/read-package-json/read-
json.js:113:5)
4 verbose stack     at /usr/share/npm/node_modules/read-package-json/read-json.j
s:232:12
4 verbose stack     at /usr/share/npm/node_modules/graceful-fs/graceful-fs.js:76
:16
4 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3
)
5 verbose cwd /home/radio/rdio-scanner
6 error Linux 4.15.0-70-generic
7 error argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
8 error node v8.10.0
9 error npm  v3.5.2
10 error missing script: build
11 error If you need help, you may report this error at:
11 error     <https://github.com/npm/npm/issues>
12 verbose exit [ 1, true ]
radio@rdio:~/rdio-scanner$ more npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'run', 'build' ]
2 info using npm@3.5.2
3 info using node@v8.10.0
4 verbose stack Error: missing script: build
4 verbose stack     at run (/usr/share/npm/lib/run-script.js:147:19)
4 verbose stack     at /usr/share/npm/lib/run-script.js:57:5
4 verbose stack     at /usr/share/npm/node_modules/read-package-json/read-json.js:345:5
4 verbose stack     at checkBinReferences_ (/usr/share/npm/node_modules/read-package-json/read-json.js:309:45)
4 verbose stack     at final (/usr/share/npm/node_modules/read-package-json/read-json.js:343:3)
4 verbose stack     at then (/usr/share/npm/node_modules/read-package-json/read-json.js:113:5)
4 verbose stack     at /usr/share/npm/node_modules/read-package-json/read-json.js:232:12
4 verbose stack     at /usr/share/npm/node_modules/graceful-fs/graceful-fs.js:76:16
4 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete] (fs.js:511:3)
5 verbose cwd /home/radio/rdio-scanner
6 error Linux 4.15.0-70-generic
7 error argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
8 error node v8.10.0
9 error npm  v3.5.2
10 error missing script: build
11 error If you need help, you may report this error at:
11 error     <https://github.com/npm/npm/issues>
12 verbose exit [ 1, true ]
radio@rdio:~/rdio-scanner$

kswanson71, 2019-11-27:
Looks like a missing build script file from what I can determine

chuot, 2019-11-27:
Where you in the 'client' subfolder when you tried 'npm run build'? You 
have to.

kswanson71, 2019-11-27:
Ah ok... 

radio@rdio:~/rdio-scanner/client$ npm run build

> rdio-scanner-client@2.1.0 build /home/radio/rdio-scanner/client
> ng build --cross-origin use-credentials --prod $*

You are running version v8.10.0 of Node.js, which is not supported by Angular CLI 8.0+.
The official Node.js version that is supported is 10.9 or greater.

Please visit https://nodejs.org/en/ to find instructions on how to update Node.js.

npm ERR! Linux 4.15.0-70-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "build"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! rdio-scanner-client@2.1.0 build: `ng build --cross-origin use-credentials --prod $*`
npm ERR! Exit status 3
npm ERR!
npm ERR! Failed at the rdio-scanner-client@2.1.0 build script 'ng build --cross-origin use-credentials --prod $*'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the rdio-scanner-client package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     ng build --cross-origin use-credentials --prod $*
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs rdio-scanner-client
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls rdio-scanner-client
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/radio/rdio-scanner/client/npm-debug.log

This is a good exercise - looks like I need to update node.js

chuot, 2019-11-27:
Keep me posted when you will have installed a newer version of Node.js.

I will update the README.md accordingly in regards to that.

kswanson71, 2019-11-27:
It's updated and I'm running npm run build in the client directory now

kswanson71, 2019-11-27:
Ok, so after running out of memory on the Hyper-V VM (2012 R2) because I think that the Gen 2 VM dynamic memory wasn't working properly...  I set it to 4GB and re-ran -

Got this result - 

radio@rdio:~/rdio-scanner/client$ npm run build

> rdio-scanner-client@2.1.0 build /home/radio/rdio-scanner/client
> ng build --cross-origin use-credentials --prod $*

Generating ES5 bundles for differential loading...
ES5 bundle generation complete.

chunk {0} runtime-es2015.edb2fcf2778e7bf1d426.js (runtime) 1.45 kB [entry] [rendered]
chunk {0} runtime-es5.edb2fcf2778e7bf1d426.js (runtime) 1.45 kB [entry] [rendered]
chunk {2} polyfills-es2015.25ff13674b3056b14f77.js (polyfills) 35.9 kB [initial] [rendered]
chunk {3} polyfills-es5.fff98669e9d96cc8f205.js (polyfills-es5) 124 kB [initial] [rendered]
chunk {1} main-es2015.90dbbb5adb7d4e713d8d.js (main) 836 kB [initial] [rendered]
chunk {1} main-es5.90dbbb5adb7d4e713d8d.js (main) 935 kB [initial] [rendered]
chunk {4} styles.eada428b3e3a2f80ecc3.css (styles) 60.9 kB [initial] [rendered]
Date: 2019-11-27T02:27:26.691Z - Hash: 8b169170ced80d8ea5c8 - Time: 118965ms

kswanson71, 2019-11-27:
So it looks like it built properly.  Are there other components from a server side that I need to build as well?

There are probably a few lessons learned when running this on a VM that I can determine from this exercise. :-)

chuot, 2019-11-27:
Now back up a folder back to rdio-scanner and type 'node run.js'

kswanson71, 2019-11-27:
Ok, looks like it got a little further -

radio@rdio:~/rdio-scanner$ node run.js
Creating SQLITE database at /home/radio/rdio-scanner/server/database.sqlite...child_process.js:660
    throw err;
    ^

Error: Command failed: npm run migrate
    at checkExecSyncError (child_process.js:621:11)
    at Object.execSync (child_process.js:657:15)
    at Object.<anonymous> (/home/radio/rdio-scanner/run.js:66:22)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11 {
  status: 1,
  signal: null,
  output: [ null, null, null ],
  pid: 1759,
  stdout: null,
  stderr: null
}

chuot, 2019-11-27:
> at Object. (/home/radio/rdio-scanner/run.js:66:22)

Go to the server subfolder and 'npm start'. What does the output tells?

kswanson71, 2019-11-27:
Looks like sqlite3 is missing?  Could that have been from the initial error thrown when node.js was not up-to-date?

radio@rdio:~/rdio-scanner/server$ npm start

> rdio-scanner-server@2.1.0 start /home/radio/rdio-scanner/server
> nodemon index

[nodemon] 2.0.1
[nodemon] to restart at any time, enter `rs`
[nodemon] watching dir(s): *.*
[nodemon] watching extensions: js,mjs,json
[nodemon] starting `node index index.js`
/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81
        throw new Error(`Please install ${moduleName} package manually`);
        ^

Error: Please install sqlite3 package manually
    at ConnectionManager._loadDialectModule (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81:15)
    at new ConnectionManager (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/connection-manager.js:22:21)
    at new SqliteDialect (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/index.js:14:30)
    at new Sequelize (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/sequelize.js:324:20)
    at Object.<anonymous> (/home/radio/rdio-scanner/server/models/index.js:16:19)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/radio/rdio-scanner/server/index.js:16:15)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11
[nodemon] app crashed - waiting for file changes before starting...

chuot, 2019-11-27:
> at Object. (/home/radio/rdio-scanner/server/models/index.js:16:19)
> at Object. (/home/radio/rdio-scanner/server/index.js:16:15)

Within the same subfolder (server), type 'npm install', the try again to start.

(I'm going off for tonight)

kswanson71, 2019-11-27:
Here's the output... I'm also spinning up another VM to try the standard installation approach with properly updated version of node.js..

Thank you for the assistance!

radio@rdio:~/rdio-scanner$ cd server
radio@rdio:~/rdio-scanner/server$ npm install
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 15 packages from 5 contributors, removed 27 packages, updated 22 packages, moved 3 packages and audited 2154 packages in 15.048s
found 0 vulnerabilities

radio@rdio:~/rdio-scanner/server$ npm start

> rdio-scanner-server@2.1.0 start /home/radio/rdio-scanner/server
> nodemon index

[nodemon] 2.0.1
[nodemon] to restart at any time, enter `rs`
[nodemon] watching dir(s): *.*
[nodemon] watching extensions: js,mjs,json
[nodemon] starting `node index index.js`
/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81
        throw new Error(`Please install ${moduleName} package manually`);
        ^

Error: Please install sqlite3 package manually
    at ConnectionManager._loadDialectModule (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81:15)
    at new ConnectionManager (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/connection-manager.js:22:21)
    at new SqliteDialect (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/index.js:14:30)
    at new Sequelize (/home/radio/rdio-scanner/server/node_modules/sequelize/lib/sequelize.js:324:20)
    at Object.<anonymous> (/home/radio/rdio-scanner/server/models/index.js:16:19)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/radio/rdio-scanner/server/index.js:16:15)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11
[nodemon] app crashed - waiting for file changes before starting...

kswanson71, 2019-11-27:
Ok, the second VM I spun up with a minimum of 4GB RAM along with ensuring that node.js is updated to the most current, stable version successfully built.

I might suggest that when end users build, they ensure that the following steps are followed:

I am using Ubuntu Server 18.04.3 LTS

1. Ensure that the OS is fully updated (sudo apt update / sudo apt upgrade)
2. Ensure that git is installed (sudo apt install git)
3. Ensure that node.js and npm are installed (sudo apt install nodejs / sudo apt install npm)
4. Ensure that node.js is updated 

sudo npm cache clean -f
sudo npm install -g n
sudo n stable

5. Continue with cloning the repository per the existing instructions
6. Change directory to rdio-scanner
7. Execute node run.js

I was able to navigate to the website!

**_Looks great by the way.  Love this interface._**  

I'll be configuring it with a trunk-recorder system that I have piping audio into Zello but that's a job for tomorrow.w

closed 2019-11-27 by kswanson71

chuot, 2019-11-27:
>    1. Continue with cloning the repository per the existing instructions
>    2. Change directory to rdio-scanner
>    3. Execute node run.js

Thank you very much for sharing your experience, the problems you have
encountered should no longer be a thing now that I have updated the
README.md file according to your advice. I also increased the verbosity of
the log when you start / update the application. Thus, in case of problems,
the cause will show up directly on the screen.

Thanks also for your comment regarding the appearance of this progressive
interface, it is really appreciated. Do not hesitate to STAR this
repository and also join us on Gitter to share your experiences.