chuot/rdio-scanner #7: connect ENOENT

jacygamer, 2019-12-03:
Got rdio_scanner install and running when I try to upload talkgroups using mariadb 
.env
DB_HOST=localhost
DB_PORT=3306
DB_DIALECT=mariadb
DB_NAME=rdio_scanner
DB_USER=rdio_scanner
DB_PASS=password
DB_STORAGE=database.sqlite

NODE_ENV=production
NODE_HOST=localhost
NODE_PORT=3000

RDIO_APIKEYS=["*****"]
RDIO_PRUNEDAYS=7

I get "connect ENOENT password"
can someone tell me how to diagnose why

chuot, 2019-12-03:
Try changing `DB_HOST=localhost` to `DB_HOST=127.0.0.1`. It looks like the `Sequelize` library cannot connect to your mariadb server.

Also, why are you using mariadb instead of the default sqlite dialect?

jacygamer, 2019-12-03:
working changed to sqlite

closed 2019-12-03 by jacygamer