chuot/rdio-scanner #8: api invalid for system

jacygamer, 2019-12-06:
Fresh install lubuntu 18.04 in virtualbox on Windows 10 I changed the key in the upload-system.sh and upload-call.sh as for the key that was in the .env when I run upload-system.sh I get the error "Api invalid for for system 1" and any other systems I try to add

chuot, 2019-12-06:
Is there an error message on the server-side console?

Have you changed the `server/.env` file in any way after it was automatically created when the software was first launched? Does its content look like this:

```bash
DB_DIALECT=sqlite
DB_STORAGE=database.sqlite

NODE_ENV=production
NODE_HOST=0.0.0.0
NODE_PORT=3000

RDIO_APIKEYS=["b29eb8b9-9bcd-4e6e-bb4f-d244ada12736"]
RDIO_PRUNEDAYS=7
```

Now, the `API key` that you need to put in your download scripts is not the entire JSON parser chain, but only the key itself, like this (no square brackets, no quotes):

```bash
#!/bin/bash

basename="${2%.*}"

nice -n 19 fdkaac -o "${basename}.m4a" -m3 -S "${basename}.wav"

curl -s http://127.0.0.1:3000/api/trunk-recorder-call-upload \
     -F "audio=@${basename}.m4a;type=audio/aac" \
     -F "key=b29eb8b9-9bcd-4e6e-bb4f-d244ada12736" \
     -F "meta=@${basename}.json;type=application/json" \
     -F "system=${1:-0}"

rm -f ${basename}.*
```

Also remember to restart the server when you change a value in `server/.env`.

jacygamer, 2019-12-06:
stupid me I forgot to change the web address in the script to the actual address I'm using

closed 2019-12-06 by jacygamer

chuot, 2019-12-06:
Please give a star if you like.