chuot/rdio-scanner #9: Is there a way to remove a system once it is uploaded or to modify an existing system?

kswanson71, 2019-12-09:
Absolutely awesome application!!

I was wondering if there is a way to either delete a system from the database or to overwrite a system with new data.

One thing that I have noticed in running multiple trunk-recorders into the interface is that for very large systems, ensuring arrangement of TG selection seems to be pulled, in order from the database of talkgroups imported through the "upload-systems.sh" script.

I'm importing from a local trunk-recorder instance and also one from several states away (Georgia and Ohio) into a single interface.  The number of talkgroups is a little overwhelming. :-)

Once again, awesome interface!!

chuot, 2019-12-09:
Thank you very much for your kind words.

To modify an existing system, upload the new version of the TALKGROUP.CSV
file over the same SYSTEM_ID.

To delete a system, upload an empty file instead of TALKGROUP.CSV. For my
part I created a file with the name of EMPTY.CSV that I use for that.

kswanson71, 2019-12-09:
Worked perfectly!!

Thank you again for your help and such an awesome tool!

Kevin Swanson

closed 2019-12-09 by kswanson71