chuot/rdio-scanner #11: Enhancement: UID aliases?

ponderbox, 2020-01-17:
Trunk-player (another trunk-recorder front end) allows users to maintain unit aliases for display. Is that something you could add? Perhaps another csv file that we can upload via script?

Loving this project!

chuot, 2020-01-17:
I'm pretty busy right now working on another project. I will try to add this feature when I have more free time.

In the meantime, could you attach a sample CSV file so that I can get an idea of ​​your needs.

Please give a star if you do like this project.

ponderbox, 2020-01-17:
Just something like UID,Description like this:
101,Rescue 1
303,Engine 10
1302030,Car 302

The upload process would probably need to include the system number (just like the upload-systems.sh)

chuot, 2020-01-27:
Feature available in version 3.0

closed 2020-01-27 by chuot