chuot/rdio-scanner #12: Download support

bettse, 2020-01-25:
In case its of any interest to anyone, I hacked together download support.  I've pushed up my changes to [a fork on gitlab](https://gitlab.com/bettse/rdio-scanner).  the two relevant commits are 

html: https://gitlab.com/bettse/rdio-scanner/commit/ce05a5b7ecbdf467b1fb5d4df16b3b6c262bb960
js: https://gitlab.com/bettse/rdio-scanner/commit/57a92a74faaeb06baa1e1f394a10d4c89be3ce1a

chuot, 2020-01-25:
It is planned to add a similar feature in the upcoming version 3.0, along with alias support and other things. I'm about 50% to completion. I'll push the code from a private branch when it will be ready. Thanks for your interest.

chuot, 2020-01-27:
Feature available in version 3.0

closed 2020-01-27 by chuot