chuot/rdio-scanner #14: Higher performing database engine?

kswanson71, 2020-03-21:
Is there a way to use an external database server that could search for calls faster?  I'm noticing that anything greater than 7 days on a busy system hangs.

daryljones, 2020-03-21:
I use MySQL with rdio-scanner and it works great!  The database has bout 40 GB of data and searches are very fast.

kswanson71, 2020-03-21:
Any specific configuration steps you could provide?  @daryljones  TIA!

daryljones, 2020-03-21:
Info is in the documentation.  The config info goes in your .env file.

DB_HOST=127.0.0.1
DB_PORT=3306
DB_DIALECT=mariadb
DB_NAME=rdio_scanner
DB_USER=xxx
DB_PASS=xxx

Here's the sql to create the schema.

CREATE TABLE IF NOT EXISTS rdioScannerSystems (

`id` INT NOT NULL auto_increment PRIMARY KEY,
`createdAt` DATETIME NOT NULL,
`updatedAt` DATETIME NOT NULL,
`name` VARCHAR(255) NOT NULL,
`system` INT NOT NULL,
`talkgroups` VARCHAR(1024) NOT NULL
)

ENGINE=InnoDB;

kswanson71, 2020-03-22:
@daryljones - 
Is there a similar schema for the rdioScannerCalls table?

daryljones, 2020-03-22:
Try running update.js after configuring  MySQL or MariaDB.  The database needs to exist. I think update.js will run a migration script to create the tables.  I don't remember for sure.

kswanson71, 2020-03-22:
@daryljones -

I took portions of what you suggested and installed sqlite3 to pull the schema from the ~/rdio-scanner/server/database.sqlite file.  

All of the JSON or VARCHAR(1024) references require shifting to LONGTEXT.

CREATE TABLE `SequelizeMeta` (`name` VARCHAR(255) NOT NULL UNIQUE PRIMARY KEY);
CREATE TABLE `rdioScannerSystems` (`id` INTEGER PRIMARY KEY AUTOINCREMENT, `createdAt` DATETIME NOT NULL, `updatedAt` DATETIME NOT NULL, `name` VARCHAR(255) NOT NULL, `system` INTEGER NOT NULL, `talkgroups` JSON NOT NULL DEFAULT '[]');
CREATE TABLE sqlite_sequence(name,seq);
CREATE UNIQUE INDEX `rdio_scanner_systems_system` ON `rdioScannerSystems` (`system`);
CREATE TABLE `rdioScannerCalls` (`id` INTEGER PRIMARY KEY AUTOINCREMENT, `createdAt` DATETIME NOT NULL, `updatedAt` DATETIME NOT NULL, `audio` LONGBLOB NOT NULL, `emergency` TINYINT(1) NOT NULL, `freq` INTEGER NOT NULL, `freqList` JSON NOT NULL, `startTime` DATETIME NOT NULL, `stopTime` DATETIME NOT NULL, `srcList` JSON NOT NULL, `system` INTEGER NOT NULL, `talkgroup` INTEGER NOT NULL);
CREATE INDEX `rdio_scanner_calls_start_time` ON `rdioScannerCalls` (`startTime`);
CREATE INDEX `rdio_scanner_calls_system` ON `rdioScannerCalls` (`system`);
CREATE INDEX `rdio_scanner_calls_system_talkgroup` ON `rdioScannerCalls` (`system`, `talkgroup`);

There are also some database account and permissioning elements to consider.  Separating the database to another server adds another measure of complexity and configuration for the mysql database instance to accept remote connections.

It's working now.  Looking forward to faster searches!  Thanks for your assistance!

closed 2020-03-22 by kswanson71