chuot/rdio-scanner #15: Iphone or mobile Audio

n2vsb, 2020-04-11:
Hello - Great work on this program! I am enjoying it! I was curious if you have a fix or might be thinking about adding some "background audio" for when a mobile device is locked? Currently, when I use my Iphone X to listen to my scanner, it needs to be "awake", or not locked or the audio won't play. Once Iphone is unlocked it works fine.

chuot, 2020-04-11:
I know, it's a pain!

I also suffer from the same problem on my Samsung S9 Plus, but I managed to work around this problem by locking the process in memory. I don't know if this feature exists on iOS devices (I don't own any Apple products these days).

You haven't mentioned the browser you are using, have you tried it with Google Chrome?

It may also be useful to add an Rdio Scanner icon to your home screen from your browser, then use this icon to launch Rdio Scanner.

It seems that iOS 13 has a lot of problems with the web audio context ...

chuot, 2020-04-13:
Apple needs to improve its PWA supports. Ref: https://blog.prototyp.digital/what-we-learned-about-pwas-and-audio-playback/

closed 2020-04-13 by chuot

bettse, 2020-12-29:
I wrote an iOS app that uses rdio-scanner's graphql API, then added push notification to rdio-scanner: https://ericbetts.org/projects/wlkie-tlkie

Likely too technical a solution for most people, but I wanted to share in case someone else wants to make something real of it.