chuot/rdio-scanner #18: Not finding trunk-recorder json metafiles

waterbwuk, 2020-05-20:
I've got trunk-recorder 3.2 set up and working fine. I'm trying to get rdio-scanner set up and I'm running into an issue. It seems like it may be looking for the metadata file too soon? rdio-scanner is finding the .wav file, and it even _sometimes_ works with the json metadata, but it fails more often than not even though the file is being written. Here is the output:

```
signal:~/rdio-scanner$ node run.js
Server is running at http://0.0.0.0:3000/
File not found: /tmp/capture/Asheville/2020/5/20/10201-1589992672_856762500.json
^C
signal:~/rdio-scanner$ ls -l /tmp/capture/Asheville/2020/5/20/10201-1589992672_856762500.json
-rw-r--r-- 1 waterbwuk waterbwuk 386 May 20 12:38 /tmp/capture/Asheville/2020/5/20/10201-1589992672_856762500.json
signal:~/rdio-scanner$
```

**My rdio-scanner config:**

```
{
    "nodejs": {
        "env": "production",
        "host": "0.0.0.0",
        "port": 3000
    },
    "sequelize": {
        "dialect": "sqlite",
        "host": "localhost",
        "name": "rdio_scanner",
        "password": "[redacted]",
        "port": 3306,
        "user": "rdio_scanner",
        "storage": "database.sqlite"
    },
    "rdioScanner": {
        "access": null,
        "allowDownload": true,
        "apiKeys": "[redacted]",
        "dirWatch": [
            {
                "deleteAfter": false,
                "directory": "/tmp/capture/Asheville",
                "extension": "wav",
                "system": 1,
                "talkgroup": ".*\\/(\\d+)-.*",
                "type": "trunk-recorder"
            }
        ],
        "pruneDays": 7,
        "systems": [
            {
                "id": 1,
                "label": "Asheville P25",
                "led": "red",
                "talkgroups": [
                    {
                        "id": 10201,
                        "label": "APD 1",
                        "name": "APD 1",
                        "tag": "APD 1",
                        "group": "Asheville Police"
                    },
                    {
                        "id": 10202,
                        "label": "APD 2",
                        "name": "APD 2",
                        "tag": "APD 2",
                        "group": "Asheville Police"
                    },
                    {
                        "id": 10203,
                        "label": "APD 3",
                        "name": "APD 3",
                        "tag": "APD 3",
                        "group": "Asheville Police"
                    },
                    {
                        "id": 10208,
                        "label": "APD TAC 5",
                        "name": "APD TAC 5",
                        "tag": "APD TAC 5",
                        "group": "Asheville Police"
                    }
                ]
            }
        ],
        "useGroup": true,
        "useLed": true
    }
}
```

**trunk-recorder config:**

```
{
	"sources": [{
		"center": 857625000,
		"rate": 2400000,
		"gain": 327,
		"ppm": 0,
		"lnaGain": 222,
		"digitalRecorders": 3,
		"modulation": "qpsk",
		"driver": "osmosdr",
		"device": "rtl=00000001"
	},{
		"center": 860375000,
		"rate": 2400000,
		"gain": 327,
		"ppm": 0,
		"lnaGain": 222,
		"digitalRecorders": 2,
		"modulation": "qpsk",
		"driver": "osmosdr",
		"device": "rtl=00000002"
	}],
	"systems": [{
		"control_channels": [857987500],
		"type": "p25",
		"shortName": "Asheville",
		"recordUnknown": "true",
		"talkgroupsFile": "/opt/recorder/ashevillep25.csv",
		"callLog": "true",
		"audioArchive": "true",
		"delayCreateOutput": "true",
		"uploadScript": "encode_upload.sh"
	}],
	"controlWarnRate": 5,
	"captureDir": "/tmp/capture",
	"frequencyFormat": "mhz",
	"defaultMode": "digital"
}
```

chuot, 2020-05-20:
Someone has reported that issue already over Gitter. His solution was to put a sleep 2 seconds into the upload script. However, this is more a patch than a solution.

Some new commits on Trunk Recorder have broken something when the upload script is being called by Trunk Recorder. The upload script should be executed only when both the WAV and JSON files have been created, not before.

Maybe you could open an issue on Trunk Recorder's GitHub page, but for now, try adding a `sleep 2` in your upload script to have it fixed quickly.

waterbwuk, 2020-05-20:
Yea, I tried the sleep solution by putting one in server/lib/rdio-scanner/dir-watch.js and I still couldn't get it to work exactly right. I did find something that seems to work, though.

I am also streaming to broadcastify and convert the wav to mp3 in my upload script, so I changed the rdio-scanner extension to watch for mp3 files instead of wav and it seems to work fine now (so far).

chuot, 2020-05-20:
But your problem is with the upload script. Do I understand that you also tried with the DirWatch alternative? If so then yes, I should wait a bit more to ensure that the JSON has also been created. I'll commit a fix in regard to that. Thanks for reporting.

waterbwuk, 2020-05-20:
Correct. I tried with adding a sleep in dirwatch using setTimeout, but I don't mess with javascript at all so who knows, I may have done it wrong.

chuot, 2020-05-20:
I just commited a fix. Please try it and report back.

waterbwuk, 2020-05-20:
I ran `node update.js` and it looks like everything pulled and updated correctly. I changed my rdio-scanner config file back to wav instead of mp3 and it's still not finding the json files. All of the configuration files and scripts I am using are mostly copied and pasted from here: https://wiki.radioreference.com/index.php/Streaming_with_Trunk_Recorder_and_Liquidsoap

chuot, 2020-05-20:
I was able to reproduce. Sometimes 2 seconds isn't enough. I will rethink the logic and will then commit a fix.

chuot, 2020-05-20:
Ok, new logic in place and commited. Give it a try.

waterbwuk, 2020-05-20:
Seems to be working perfectly now. Thanks!

chuot, 2020-05-20:
Happy Rdio scanning !

closed 2020-05-20 by chuot