chuot/rdio-scanner #19: Failing to parse SDRTrunk files

waterbwuk, 2020-05-31:
I'm trying to set up rdio-scanner with SDRTrunk and it is not importing when .wav files are being written. From looking at server/lib/rdio-scanner/dir-watch.js I think the problem might be at line 139:

```
        if (file.ext === '.mbe') {
            const audioFile = path.resolve(file.dir, `${file.name}.${dirWatch.extension}`);

            if (fs.existsSync(audioFile)) {
```

When looking at my recordings folder I see files such as:

```
20200531003853_856762500_1_P10201_224357.mbe
20200531_003853Asheville_Buncombe_TRAFFIC__TO_10201_FROM_224357.wav
```

I'm assuming it is taking the base name of the .mbe file and looking for the .wav file (I specified wav in config.json), which isn't going to match. It seems like it would be looking for 20200531003853_856762500_1_P10201_224357.wav instead.

There's nothing to catch the if statement for fs.existsSync failing, therefor it's not printing out an error.

My config.json:

```
{
  "sequelize": {
    "dialect": "sqlite",
    "host": "localhost",
    "name": "rdio_scanner",
    "password": "[removed]",
    "port": 3306,
    "user": "rdio_scanner",
    "storage": "database.sqlite"
  },
  "rdioScanner": {
    "access": null,
    "allowDownload": true,
    "dirWatch": [
      {
        "deleteAfter": true,
        "directory": "/home/waterbwuk/SDRTrunk/recordings",
        "extension": "wav",
        "system": 1,
        "type": "sdrtrunk"
      }
    ],
    "pruneDays": 7,
    "systems": [
      {
        "id": 1,
        "label": "ASHEVILLE P25",
        "led": "green",
        "talkgroups": [
          {
            "id": 10201,
            "group": "POLICE",
            "label": "APD 1",
            "led": "green",
            "name": "APD 1",
            "tag": "APD DISPATCH"
          },
          {
            "id": 10202,
            "group": "POLICE",
            "label": "APD 2",
            "led": "green",
            "name": "APD 2",
            "tag": "APD 2"
          },
          {
            "id": 10862,
            "group": "COLLEGES",
            "label": "UNC PD",
            "led": "green",
            "name": "UNC PD",
            "tag": "UNC POLICE"
          }
        ]
      }
    ],
    "useGroup": true
  }
}
```

waterbwuk, 2020-05-31:
Don't know if this helps any but I added in some debug statements to make sure of what it is looking for:

```
waterbwuk@signal:~/Desktop/rdio-scanner$ node run.js
Server is running at http://0.0.0.0:3000/
importSdrTrunkType called for /home/waterbwuk/SDRTrunk/recordings/20200531122153_856762500_1_10202_225305.mbe
About to look for audioFile: 20200531122153_856762500_1_10202_225305.wav
audioFile: /home/waterbwuk/SDRTrunk/recordings/20200531122153_856762500_1_10202_225305.wav
Error loading audioFile: /home/waterbwuk/SDRTrunk/recordings/20200531122153_856762500_1_10202_225305.wav
importSdrTrunkType called for /home/waterbwuk/SDRTrunk/recordings/20200531_122154Asheville_Buncombe_TRAFFIC__TO_10202_FROM_225305.wav
importSdrTrunkType called for /home/waterbwuk/SDRTrunk/recordings/20200531122154_856762500_2_10202.mbe
About to look for audioFile: 20200531122154_856762500_2_10202.wav
audioFile: /home/waterbwuk/SDRTrunk/recordings/20200531122154_856762500_2_10202.wav
Error loading audioFile: /home/waterbwuk/SDRTrunk/recordings/20200531122154_856762500_2_10202.wav
importSdrTrunkType called for /home/waterbwuk/SDRTrunk/recordings/20200531_122155Asheville_Buncombe_TRAFFIC__TO_10202.wav
^C
waterbwuk@signal:~/Desktop/rdio-scanner$ ls /home/waterbwuk/SDRTrunk/recordings
20200531122153_856762500_1_10202_225305.mbe  20200531122154_856762500_2_10202.mbe  20200531_122154Asheville_Buncombe_TRAFFIC__TO_10202_FROM_225305.wav  20200531_122155Asheville_Buncombe_TRAFFIC__TO_10202.wav
waterbwuk@signal:~/Desktop/rdio-scanner$
```

chuot, 2020-05-31:
If the metafile doesn't have the same name as for the wav file, the logic I'm using to import those files will never work.

However, I see that with SDRTrunk 0.5.0 you can tell to save audio files from the alias tab, which then create a MP3 with mostly all the meta information from the filename itself.

I will then rework the import for SDRTrunk to make it injest all the meta informations from the filename itself.

waterbwuk, 2020-06-01:
In the meantime, I managed to get this working by changing the dirWatch type from "sdrtrunk" to "default". I then added in a talkgroup entry (the P_ part for when they use patches):

```
    "dirWatch": [
      {
        "deleteAfter": true,
        "directory": "/home/waterbwuk/SDRTrunk/recordings",
        "extension": "wav",
        "system": 1,
        "talkgroup": ".*TRAFFIC__TO_[P]?[_]?(\\d+).*",
        "type": "default"
      }
    ],
```

chuot, 2020-06-01:
Yep, that is where I'm heading too. I need to add the unit id to the code.

I'm also lurking a way to specify a mask, something  like "%S_TRAFFIC_TO_%T_FROM_%U".

It should be there for v4.2.

chuot, 2020-06-01:
V4.2 just released with working `dirWatch.mask`.

SDRTrunk should now use dirWatch.mask with dirWatch.type="default" per `docs/config/dir-watch.md` instructions.

closed 2020-06-01 by chuot