chuot/rdio-scanner #20: Failing to load recordings from trunk-recorder "DirWatch: Error loading file"

cmnybo, 2020-06-03:
I am trying to use this with trunk-recorder and every time a new recording is added, rdio-scanner fails to load the json file. None of the recordings are being loaded and nothing has been written to the sqlite database.

`DirWatch: Error loading file /home/cody/Public/Trunk-Recorder/sers/2020/6/2/1680-1591167062_853362500.json`

Here is my config file and one of the trunk-recorder json files.
[config.json](https://github.com/chuot/rdio-scanner/files/4721439/config.json.txt)
[1680-1591167062_853362500.json](https://github.com/chuot/rdio-scanner/files/4721446/1680-1591167062_853362500.json.txt)

chuot, 2020-06-03:
In your trunk-recorder's config, does audioArchive=true and callLog=true? It looks like one of the json/wav file has vanished before it's being imported. Double check the audioArchive/callLog and come back. I may change the code a bit to give better indications in regards to the error.

cmnybo, 2020-06-03:
OK, I set audioArchive and callLog to true, but I still get the same error.

Here is the config file for trunk-recorder.
[config.json](https://github.com/chuot/rdio-scanner/files/4725724/config.json.txt)

chuot, 2020-06-03:
Are trunk-recorder and rdio-scanner running on the same machine, or the audio_files folder is nfs/cifs mounted? If so, you should add `dirWatch.usePolling=true`. Also, the `dirWatch.delay` isn't used with `dirWatch.type="trunk-recorder"`, so you can remove it.

Can you try to manually paste at the same time one wav file with its json file into the dirWatch directory and see if they're imported by Rdio Scanner?

chuot, 2020-06-03:
I just commited changes to the dirWatch which will give more precise error message. Do an update before trying the new test.

daryljones, 2020-06-03:
What's the preferred method of loading files into the rdio-scanner
database?  I'm still using the original upload-calls.sh script.

chuot, 2020-06-03:
@daryljones If Rdio Scanner and Trunk Recorder are on the same machine, both methods are valid. If Rdio Scanner and Trunk Recorder aren't on the same machine, then you should stick with the uploadScript.

It is also possible to have many instances of Rdio Scanner as feeders using `dirWatch` and `downstream` each instance to your main Rdio Scanner instance. This setup is recommended when you have distant feeders, say in defferent cities/locations.

cmnybo, 2020-06-03:
OK, I updated rdio-scanner and removed the dirWatch delay from the config file. I started pasting the recordings back into the watch folder and they were all imported. I started trunk-recorder again and now it's working. The new recordings are being imported and playing correctly. trunk-recorder and rdio-scanner are both running on the same machine.

closed 2020-06-04 by chuot