chuot/rdio-scanner #23: Documentation or setup issues

charlie-foxtrot, 2020-06-28:
I ran into some problems when setting up for the first time:

1. When running for the first time, https://github.com/chuot/rdio-scanner#quick-start shows a default configuration will be created, provides an API token, and creates a SQLITE database.  That didn't work for me, instead it stated with some default config but there was no `server/config.json` created.  I was able to copy one of the examples into `server/config.json` and create a db using `npm run migrate`
2. https://github.com/chuot/rdio-scanner/blob/master/docs/config/dir-watch.md shows an example that sets `talkgroups` to a single value, however when I did that all incoming files said no talk group found.  When I removed the `s` (ie `talkgroup`) then it seemed to work, except for below
3. When I setup `dirWatch` to have multiple entries with the same directory but different masks, a recording matches every entry.  Here is the part of the config:
```
            {
                "deleteAfter": "false", 
                "directory": "/recordings", 
                "extension": "mp3", 
                "frequency": 158737500, 
                "mask": "Agency_CoOp-Tactical-VTAC13_#DATE_#TIME", 
                "system": 50, 
                "talkgroup": 873750, 
                "type": "default"
            }, 
            {
                "deleteAfter": "false", 
                "directory": "/recordings", 
                "extension": "mp3", 
                "frequency": 159472500, 
                "mask": "Agency_CoOp-Tactical-VTAC14_#DATE_#TIME", 
                "system": 50, 
                "talkgroup": 947250, 
                "type": "default"
            }
```
and this is what happens when a file shows up:
```
NewCall: system=50 talkgroup=947250 file=Agency_CoOp-Tactical-VTAC13_20200628_104915.m4a Success
NewCall: system=50 talkgroup=873750 file=Agency_CoOp-Tactical-VTAC13_20200628_104915.m4a Success
DirWatch: Unable to delete /recordings/Agency_CoOp-Tactical-VTAC13_20200628_104915.mp3
```
4. Setting `deleteAfter` to `false` still deletes the file (see above).

chuot, 2020-06-28:
Thanks @charlie-foxtrot for reporting these issues. I'm currently working on v4.3 where I will test more thoroughly the dirWatch feature. I'll make sure that these bugs will be fixed. v4.3 will also take care of the migration scripts... There's a lot of code refactoring ahead.

Stay tuned!

chuot, 2020-07-02:
@charlie-foxtrot While v4.3 is taking more time to produce than I first anticipated, I just released a quick fix for dirWatch.

Note that this fix will ensure that a directory isn't declared twice in dirWatch. This is your case, so you need to rethink your configuration.

The fix will also ensure that dirWatch.deleteAfter is boolean and true to delete the file. Your config passes a string instead of a boolean, which was treated as a thruty value.

Since your using the same folder for your recordings, you must defined it only once in the dirWatch section. Consequently, you must rely on the dirWatch.mask for the system and talkgroup values. I would recommend to change your (rtlsdr_airband) filename_template to include embeded numerical values for system and talkgroup.

Try it like this:

```json
            {
                "deleteAfter": false, 
                "directory": "/recordings", 
                "extension": "mp3", 
                "frequency": 159472500, 
                "mask": "Agency_CoOp-Tactical-VTAC13_#SYS_#TG_#DATE_#TIME"
            }
```

However, if you do want to keep your filenames intact, then split them into subfolders and use those subfolders in your dirWatch definitions.

Note that I'm not yet closing the issue since documentations need to be change accordingly, but no more time for me to work on this right now. Version 4.3 will fix the remaining stuff.

chuot, 2020-07-11:
Version 4.3 should now resolves all these issues.

closed 2020-07-11 by chuot