chuot/rdio-scanner #29: dirWatch file extension ignored when type is trunk-recorder

k-worker, 2020-12-06:
I see that in 4.7 there was a change made to ignore the specified file extension for a dirWatch audio source when the type is trunk-recorder or sdr-trunk. This might make sense when trunk-recorder is running on the same machine as rdio-scanner, but breaks certain setups where trunk-recorder is uploading to the system rdio-scanner is on (like mine).

I use ffmpeg to encode to an m4a file on my local trunk-recorder system before uploading to my VPS hosting rdio-scanner. This way I save bandwidth, save disk space on the VPS, and don't waste CPU cycles transcoding to m4a on the VPS. However, rdio-scanner is hardcoded to look for wav files, and doesn't import any calls.

This worked before, and IMO it makes sense to let the user specify a file extension if it's not the default. Is this a use-case/setup that's intentionally not supported anymore? Or is there a preferred alternative to this kind of setup?

Here's the line in question: https://github.com/chuot/rdio-scanner/blob/master/server/lib/rdio-scanner/dir-watch.js#L101

I'd offer a PR but am a bit short on javascript experience and free time right now.

chuot, 2020-12-06:
Thanks for the report, I'll look at it tomorrow.

chuot, 2020-12-07:
It should work now as of version 4.8.3.

Thanks again for reporting the issue.

closed 2020-12-07 by chuot