chuot/rdio-scanner #32: Add keyboard shortcuts

theY4Kman, 2020-12-28:
It would be nice if some actions could be performed with the keyboard. I often use my laptop in the car to listen to my scanner, and it can be tough — as well as dangerous and possibly illegal — to drop the cursor to replay or skip or pause.

I've been playing with [ng-keyboard-shortcuts](https://github.com/omridevk/ng-keyboard-shortcuts) in my local rdio-scanner, and have found these shortcuts super useful:

 - `L` to toggle live feed
 - Spacebar to Pause
 - Left Arrow to Replay Last
 - Right Arrow to Skip/Next
 - Escape/Backspace to exit Select/Search panels

I'd also added shortcuts for the other functions, but I use them less

 - `/` to open Search panel
 - `s` to open Select TG panel
 - `h`, then `s` to Hold System
 - `h`, then `g` to Hold TG

theY4Kman, 2020-12-28:
I plan to open a standalone PR for keyboard shortcuts, but the code is currently visible intermingled in my fork: https://github.com/theY4Kman/rdio-scanner

chuot, 2020-12-28:
Thanks for the suggestion.

It's something I already do with my RC Scanner, my other product, since it's very use full when you want to configure system/talkgroup names on a real scanner. I can easilly bring that same feature to Rdio Scanner. However, no need for an extra package since it can be done with a since @HostListener.

I've already planned a new release (4.8.6) in the first week of 2021 where you should see these keyboard shortcuts. They won't be like the one you're proposing because I prefer single key shortcuts over combo shortcuts (s instead of h+s).

closed 2021-01-02 by chuot