chuot/rdio-scanner #34: Invalid API key for system

VVakko, 2021-01-07:
I installed the latest version of "Rdio Scanner" 4.9.2 under Docker. And for a long time I could not set up interaction with "Trunk Recorder" via the API. For each attempt to upload a call, I received the error "Invalid API key for system 107 talkgroup 7". I tried to re-generate the key – it didn't help.

Then it turned out that if the system and talkgroup are specified in config.json – upload is successful: "Call imported successfully". But if a new talkgroup starts transmitting – we get an error: "Invalid API key for system 107 talkgroup 7".

I attach config.json:
```
{
  "nodejs": {
    "env": "production",
    "host": "0.0.0.0",
    "port": "3000",
    "sslCA": null,
    "sslCert": null,
    "sslKey": null
  },
  "rdioScanner": {
    "access": null,
    "apiKeys": "...",
    "dirWatch": [],
    "downstreams": [],
    "options": {
      "dimmerDelay": 5000,
      "disableAudioConversion": false,
      "disableKeyboardShortcuts": false,
      "keypadBeeps": 1,
      "pruneDays": 7
    },
    "systems": [
      {
        "id": 103,
        "label": "Site: 1.3",
        "talkgroups": [
          {
            "id": 7
          },
          {
            "id": 87
          },
          {
            "id": 91
          },
          {
            "id": 102
          }
        ],
        "units": []
      },
      {
        "id": 107,
        "label": "Site: 1.7",
        "talkgroups": [
          {
            "id": 7
          },
          {
            "id": 87
          },
          {
            "id": 91
          },
          {
            "id": 102
          }
        ],
        "units": []
      }
    ]
  },
  "sequelize": {
    "database": null,
    "dialect": "sqlite",
    "host": null,
    "password": null,
    "port": null,
    "storage": "/app/data/database.sqlite",
    "username": null
  }
}
```

How can I allow uploading of all conversations regardless of whether talkgroups are presented in config.json or not?

chuot, 2021-01-07:
There's 2 conditions for the API to accept new calls. 1. the apikey must allow the system/talkgroup (rdioScanner.apiKeys). 2. the system/talkgroup must be known by Rdio Scanner (rdioScanner.systems).

It looks like the format you're using for the apikey isn't correct, but I can't be sure since it's mask in your example.

To allow an apikey for all the systems/talkgroups, try one of those:

```json
"rdioScanner": {
  "apiKeys": "your-api-key"
}
```

or

```json
"rdioScanner": {
  "apiKeys": [
    {
      "key": "your-api-key",
      "systems": "*"
    }
  ]
}
```

For support requests, please use the [Rdio Scanner/Lobby](https://gitter.im/rdio-scanner/Lobby) at Gitter.

VVakko, 2021-01-07:
Yes, I tried both of these options.

If I do config.json like this (without the block "systems"):
```
{
  "nodejs": {
    "env": "production",
    "host": "0.0.0.0",
    "port": "3000",
    "sslCA": null,
    "sslCert": null,
    "sslKey": null
  },
  "rdioScanner": {
    "access": null,
    "apiKeys": "...",
    "dirWatch": [],
    "downstreams": [],
    "options": {
      "dimmerDelay": 5000,
      "disableAudioConversion": false,
      "disableKeyboardShortcuts": false,
      "keypadBeeps": 1,
      "pruneDays": 7
    }
  },
  "sequelize": {
    "database": null,
    "dialect": "sqlite",
    "host": null,
    "password": null,
    "port": null,
    "storage": "/app/data/database.sqlite",
    "username": null
  }
}
```

Then after starting the Docker container, in the config.json automatically added the default block "systems":
```
    "systems": [
      {
        "id": 11,
        "label": "RSP25MTL1",
        "talkgroups": [
          {
            "group": "FIRE DISPATCH",
            "id": 54241,
            "label": "TDB A1",
            "name": "MRC TDB Fire Alpha 1",
            "tag": "Fire Dispatch"
          },
          {
            "group": "FIRE TAC",
            "id": 54242,
            "label": "TDB B1",
            "name": "MRC TDB Fire Bravo 1",
            "tag": "Fire Tac"
          },
```
And on any upload I get the error "Invalid API key for system xx talkgroup yy".

If in the config.json write an empty block "systems": [], then after restarting the container, the default values are not written, but the error "Invalid API key for system xx talkgroup yy" remains.

Works only one option, when in config.json presented all systems and all talkgroups.

chuot, 2021-01-07:
As said in the [docs/config.md](https://github.com/chuot/rdio-scanner/blob/master/docs/config.md):

> ## Systems - rdioScanner.systems
> 
> The heart of [Rdio Scanner](https://github.com/chuot/rdio-scanner) where all your systems and talkgroups are defined. Audio files to unknown systems / talkgroups will not be ingested.

So having systems and talkgroups well defined in `rdioScanner.config.systems` is a must for Rdio Scanner to work.

VVakko, 2021-01-07:
Okay, got it. Thanks!

I ask you to consider the option of leaving only systems as mandatory, since they are really static. In our area, the P25 trunk is in the process of being built and configured, and new groups appear quite often. In order to hear all the groups, I have now just listed all the numbers from 1 to 255 as talkgroups.
Perhaps there is an option in the future to make talkgroups optional as units?

chuot, 2021-01-08:
The new version 5.0 with the autoPopulate option should solve your issue. Please test and report back over Gitter. Thanks

closed 2021-01-08 by chuot