chuot/rdio-scanner #35: Downstream + Self-Signed SSL

adorsett, 2021-02-10:
I have a rdio-scanner instance running on host1 on the default port of 3000 using DirWatch to receive from trunk-recorder.  On host2 I have rdio-scanner running on port 8443 with a self-signed certificate (using the procedure from config.md to create).  I am using downstream from host1 to push to host2 for feed aggregation.  When SSL is enabled on host2 I get the following log entry on host1  (sender) and nothing at all logged on host2 (receiver) and the upload never completes: 

`Downstream: system=1 talkgroup=17392 file=17392-1612988568_856412500.m4a to=https://192.168.100.61:8443/ self signed certificate`

If I disable SSL on host2 and change the url to http:// for downstream in host1's config I get this log entry on host1:
`Downstream: system=1 talkgroup=17424 file=17424-1612989173_856412500.m4a to=http://192.168.100.61:3000/ Success`

and I see this on host2:
`NewCall: system=1 talkgroup=17424 file=17424-1612989173_856412500.m4a Success`

Thus with self-signed SSL disabled it works fully.

It appears that rdio-scanner does not like the self-signed cert on host1 and is not trusting it.  Do we need a way to import the cert on host1 so that it will trust the connection to host2?

chuot, 2021-02-11:
Thanks for reporting this issue. You should be good to go with the version 5.0.4 which fixes this issue. I've successfully tested a downstream link to an instance with a self-signed certificate.

closed 2021-02-11 by chuot

adorsett, 2021-02-11:
Verified the solution solved the issue.  Thanks for the prompt fix!