chuot/rdio-scanner #36: "NO LINK" with HTTPS + Self Signed Cert on IOS with Safari

adorsett, 2021-02-12:
Running IOS 14.5 Beta.  Enabled HTTPS with self signed cert (per my other issue you resolved).  I get the cert validation failure message on Safari and I proceed anyway.  The UI loads and says "NO LINK" at the very top.  I've tried both externally (through NAT+FW) and directly connected (WIFI) to the rdio-scanner host.  If I disable HTTPS it works as expected.

chuot, 2021-02-13:
I was able to reproduce your problem with this error: `Websocket network error: the operation could not be completed. (OSStatus error -9807.)`. In other words: Safari prohibits Websocket communication when using self-signed certificates. I tried with Google Chrome but it's the same. Since it's not an issue with Rdio Scanner, but more an IOS thing, I won't be able to solve this one. Can I recommend you to try with a letsencrypt.org certificate? I do and it works well.

chuot, 2021-04-04:
Closing this issue as I can't do nothing about it.

closed 2021-04-04 by chuot