chuot/rdio-scanner #38: Unable to build Rdio Scanner -Throw new Error(Please install ${moduleName} package manually);

photounit, 2021-04-04:
I'm trying to install Rdio Scanner in Ubuntu 20.04 which I have done numerous times before. When I try to install it lately to a fresh install of Ubuntu 20.04 I get a "Error: Please install sqlite3 package manually". I'm stumped as I've tried searching the web and everything I know. Any help would be appreciated.   Here is a copy of the debug log:                                                                                                                                                                                                                                              openmhz1@openmhz1:~/rdio-scanner$ DEBUG=true node run.js
Building client app...

    rdio-scanner-client@5.0.4 build /home/openmhz1/rdio-scanner/client
    ng build --base-href ./ --prod

Compiling @angular/animations : es2015 as esm2015
Compiling @angular/core : es2015 as esm2015
Compiling @angular/cdk/keycodes : es2015 as esm2015
Compiling @angular/common : es2015 as esm2015
Compiling @angular/animations/browser : es2015 as esm2015
Compiling @angular/cdk/observers : es2015 as esm2015
Compiling @angular/cdk/collections : es2015 as esm2015
Compiling @angular/cdk/platform : es2015 as esm2015
Compiling @angular/cdk/bidi : es2015 as esm2015
Compiling @angular/platform-browser : es2015 as esm2015
Compiling @angular/cdk/a11y : es2015 as esm2015
Compiling @angular/forms : es2015 as esm2015
Compiling @angular/cdk/portal : es2015 as esm2015
Compiling @angular/platform-browser/animations : es2015 as esm2015
Compiling @angular/cdk/scrolling : es2015 as esm2015
Compiling @angular/cdk/layout : es2015 as esm2015
Compiling @angular/cdk/text-field : es2015 as esm2015
Compiling @angular/common/http : es2015 as esm2015
Compiling @angular/platform-browser-dynamic : es2015 as esm2015
Compiling @angular/service-worker : es2015 as esm2015
Compiling @angular/material/core : es2015 as esm2015
Compiling @angular/cdk/overlay : es2015 as esm2015
Compiling @angular/cdk/table : es2015 as esm2015
Compiling @angular/material/button : es2015 as esm2015
Compiling @angular/material/form-field : es2015 as esm2015
Compiling @angular/material/tooltip : es2015 as esm2015
Compiling @angular/material/dialog : es2015 as esm2015
Compiling @angular/material/sort : es2015 as esm2015
Compiling @angular/material/card : es2015 as esm2015
Compiling @angular/material/select : es2015 as esm2015
Compiling @angular/material/input : es2015 as esm2015
Compiling @angular/material/icon : es2015 as esm2015
Compiling @angular/material/progress-bar : es2015 as esm2015
Compiling @angular/material/datepicker : es2015 as esm2015
Compiling @angular/material/sidenav : es2015 as esm2015
Compiling @angular/material/paginator : es2015 as esm2015
Compiling @angular/material/slide-toggle : es2015 as esm2015
Compiling @angular/material/table : es2015 as esm2015
Compiling @angular/material/toolbar : es2015 as esm2015
✔ Browser application bundle generation complete.
✔ Copying assets complete.
✔ Index html generation complete.
✔ Service worker generation complete.

Initial Chunk Files | Names | Size
main.196f2f369d8fc3a01bf5.js | main | 717.59 kB
styles.6649fa4b4334da7cd24b.css | styles | 71.83 kB
polyfills.717003aa952d9ffe40c2.js | polyfills | 35.73 kB
runtime.7b63b9fd40098a2e8207.js | runtime | 1.45 kB

                              | Initial Total | 826.59 kB

Build at: 2021-04-04T04:23:16.519Z - Hash: 24c02b25751b8a6c06e2 - Time: 51732ms
done
/home/openmhz1/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81
throw new Error(Please install ${moduleName} package manually);
^

Error: Please install sqlite3 package manually
at ConnectionManager._loadDialectModule (/home/openmhz1/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/connection-manager.js:81:15)
at new ConnectionManager (/home/openmhz1/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/connection-manager.js:24:21)
at new SqliteDialect (/home/openmhz1/rdio-scanner/server/node_modules/sequelize/lib/dialects/sqlite/index.js:15:30)
at new Sequelize (/home/openmhz1/rdio-scanner/server/node_modules/sequelize/lib/sequelize.js:340:20)
at new App (/home/openmhz1/rdio-scanner/server/index.js:185:26)
at Object.<anonymous> (/home/openmhz1/rdio-scanner/server/index.js:275:5)
at Module._compile (internal/modules/cjs/loader.js:1063:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:10

chuot, 2021-04-04:
Which version of nodejs you are using (node -v)? Also, the sqlite node package requires python to be installed. Is it installed (python -V)?

photounit, 2021-04-04:
Thanks for the response...                                                                                                                                                                                node -v  = v14.16.0

I don't have python installed but I don't remember installing that and I don't remember that as being a perquisite. I guess I will install it and try again....

photounit, 2021-04-05:
Chrystian, Well installing python was the fix....  Apparently whenever I had installed Rdio Scanner before I had python installed already. I didn't see anything in the errors or web searches of similar issues that would have led me to the lack of Python being the issue.  I have one request, can you add a complete functional config.json file to the examples? It would help to have as a reference to use in troubleshooting for formatting errors....  Thanks again for your help and for your excellent program....

chuot, 2021-04-06:
There's already an example under docs/rdio-scanner/config.json for the time being. Things will change with the upcoming v5.1, so I don't want to spent to much time on something that will change soon. I do have a complete setup I did for someone else. I will DM you the details over at gitter.

closed 2021-04-06 by chuot