chuot/rdio-scanner #40: 5.1 Upgrade Fails

kcwebby, 2021-05-18:
Using MariaDB instead of sqlite, I'm getting this error after an upgrade.

```
APP_DATA=/root/master node index.js
Running migration 20210115105958-new-v5...(node:8134) UnhandledPromiseRejectionWarning: SequelizeUniqueConstraintError: Validation error
    at Query.formatError (/root/rdio-scanner/server/node_modules/sequelize/lib/dialects/mariadb/query.js:254:16)
    at Query.run (/root/rdio-scanner/server/node_modules/sequelize/lib/dialects/mariadb/query.js:66:18)
    at processTicksAndRejections (internal/process/task_queues.js:93:5)
    at async /root/rdio-scanner/server/node_modules/sequelize/lib/sequelize.js:619:16
    at async MySQLQueryInterface.bulkInsert (/root/rdio-scanner/server/node_modules/sequelize/lib/dialects/abstract/query-interface.js:833:21)
    at async Object.up (file:///root/rdio-scanner/server/lib/rdio-scanner/migrations/20210115105958-new-v5.1-tables.js:130:17)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8134) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:8134) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

chuot, 2021-05-18:
Thanks for reporting the issue.

I did test the upgrade on a mariadb setup without any issue. Maybe I changed something afterwards that break the upgrade, or maybe it's something else. Anyway, I will retest again and will come back to you on this.

chuot, 2021-05-19:
It should work now with v5.1.1

closed 2021-05-19 by chuot