chuot/rdio-scanner #46: Can't run rdio-scanner docker installation. Error: ENOENT: no such file or directory, stat '/app/data/ca.srl',

atlex, 2021-07-09:
Hi,

I can't run the rdio-scanner. It's fresh new docker installation.
I did:

$ docker run --detach --env TZ=Europe/Minsk --name rdio-scanner --publish 3000:3000 --restart always --user $(id -u):$(id -g) --volume ~/.rdio-scanner:/app/data --volume ~/recs:/home/radio/recordings chuot/rdio-scanner:latest

$ docker restart rdio-scanner
$ docker ps

CONTAINER ID   IMAGE                       COMMAND         CREATED          STATUS                          PORTS                                       NAMES
371df2461f57   chuot/rdio-scanner:latest   "node server"   33 minutes ago   Restarting (1) 8 seconds ago                                               rdio-scanner

I can't connect to http://192.168.1.8:3000

$ tail -f /var/log/syslog

Jul  9 13:40:23 srv1 kernel: [ 2710.582582] docker0: port 2(veth0413590) entered disabled state
Jul  9 13:40:23 srv1 systemd-networkd[581]: veth0413590: Lost carrier
Jul  9 13:40:23 srv1 kernel: [ 2710.590457] vethef25ff9: renamed from eth0
Jul  9 13:40:23 srv1 networkd-dispatcher[611]: WARNING:Unknown index 128 seen, reloading interface list
Jul  9 13:40:23 srv1 systemd-networkd[581]: veth0413590: Link DOWN
Jul  9 13:40:23 srv1 kernel: [ 2710.611471] docker0: port 2(veth0413590) entered disabled state
Jul  9 13:40:23 srv1 systemd-networkd[581]: rtnl: received neighbor for link '129' we don't know about, ignoring.
Jul  9 13:40:23 srv1 kernel: [ 2710.622748] device veth0413590 left promiscuous mode
Jul  9 13:40:23 srv1 kernel: [ 2710.622758] docker0: port 2(veth0413590) entered disabled state
Jul  9 13:40:23 srv1 systemd-networkd[581]: rtnl: received neighbor for link '129' we don't know about, ignoring.
Jul  9 13:40:23 srv1 systemd[4487]: run-docker-netns-50fef7e0d196.mount: Succeeded.
Jul  9 13:40:23 srv1 systemd[1]: run-docker-netns-50fef7e0d196.mount: Succeeded.
Jul  9 13:40:23 srv1 systemd[1]: var-lib-docker-overlay2-9a86ae7d6b3c1b99f2ffb7b4c4396e6d6473d6fb540bc8969636577e53edbf7e-merged.mount: Succeeded.
Jul  9 13:40:23 srv1 systemd[4487]: var-lib-docker-overlay2-9a86ae7d6b3c1b99f2ffb7b4c4396e6d6473d6fb540bc8969636577e53edbf7e-merged.mount: Succeeded.
Jul  9 13:40:23 srv1 networkd-dispatcher[611]: ERROR:Unknown interface index 128 seen even after reload
Jul  9 13:40:23 srv1 systemd-udevd[14054]: vethef25ff9: Failed to get link config: No such device
Jul  9 13:40:23 srv1 networkd-dispatcher[611]: WARNING:Unknown index 128 seen, reloading interface list
Jul  9 13:40:23 srv1 networkd-dispatcher[611]: ERROR:Unknown interface index 128 seen even after reload
Jul  9 13:40:23 srv1 systemd-udevd[14055]: vethef25ff9: Failed to get link config: No such device

$ docker logs rdio-scanner

    at Function.Module._load (internal/modules/cjs/loader.js:790:14),
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12),
    at internal/main/run_main_module.js:17:47 {,
  errno: -2,,
  syscall: 'stat',,
  code: 'ENOENT',,
  path: '/app/data/ca.srl',
},
internal/fs/utils.js:732,
      throw err;,
      ^,
,
Error: ENOENT: no such file or directory, stat '/app/data/ca.srl',
    at Object.statSync (fs.js:1127:3),
    at internal/fs/utils.js:717:32,
    at hidden (internal/errors.js:314:14),
    at Object.rmSync (fs.js:931:13),
    at new <anonymous> (/app/server/index.js:1:71934),
    at /app/server/index.js:1:69678,
    at Object.<anonymous> (/app/server/index.js:1:74126),
    at Module._compile (internal/modules/cjs/loader.js:1085:14),
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10),
    at Module.load (internal/modules/cjs/loader.js:950:32),
    at Function.Module._load (internal/modules/cjs/loader.js:790:14),
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12),
    at internal/main/run_main_module.js:17:47 {,
  errno: -2,,
  syscall: 'stat',,
  code: 'ENOENT',,
  path: '/app/data/ca.srl',
},
internal/fs/utils.js:732,
      throw err;,
      ^,
,
Error: ENOENT: no such file or directory, stat '/app/data/ca.srl',
    at Object.statSync (fs.js:1127:3),
    at internal/fs/utils.js:717:32,
    at hidden (internal/errors.js:314:14),
    at Object.rmSync (fs.js:931:13),
    at new <anonymous> (/app/server/index.js:1:71934),
    at /app/server/index.js:1:69678,
    at Object.<anonymous> (/app/server/index.js:1:74126),
    at Module._compile (internal/modules/cjs/loader.js:1085:14),
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10),
    at Module.load (internal/modules/cjs/loader.js:950:32),
    at Function.Module._load (internal/modules/cjs/loader.js:790:14),
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12),
    at internal/main/run_main_module.js:17:47 {,
  errno: -2,,
  syscall: 'stat',,
  code: 'ENOENT',,
  path: '/app/data/ca.srl',
},
internal/fs/utils.js:732,
      throw err;,
      ^,
,
Error: ENOENT: no such file or directory, stat '/app/data/ca.srl',
    at Object.statSync (fs.js:1127:3),
    at internal/fs/utils.js:717:32,
    at hidden (internal/errors.js:314:14),
    at Object.rmSync (fs.js:931:13),
    at new <anonymous> (/app/server/index.js:1:71934),
    at /app/server/index.js:1:69678,
    at Object.<anonymous> (/app/server/index.js:1:74126),
    at Module._compile (internal/modules/cjs/loader.js:1085:14),
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10),
    at Module.load (internal/modules/cjs/loader.js:950:32),
    at Function.Module._load (internal/modules/cjs/loader.js:790:14),
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12),
    at internal/main/run_main_module.js:17:47 {,
  errno: -2,,
  syscall: 'stat',,
  code: 'ENOENT',,
  path: '/app/data/ca.srl',
},
internal/fs/utils.js:732,
      throw err;,
      ^,
,
Error: ENOENT: no such file or directory, stat '/app/data/ca.srl',
    at Object.statSync (fs.js:1127:3),
    at internal/fs/utils.js:717:32,
    at hidden (internal/errors.js:314:14),
    at Object.rmSync (fs.js:931:13),
    at new <anonymous> (/app/server/index.js:1:71934),
    at /app/server/index.js:1:69678,
    at Object.<anonymous> (/app/server/index.js:1:74126),
    at Module._compile (internal/modules/cjs/loader.js:1085:14),
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10),
    at Module.load (internal/modules/cjs/loader.js:950:32),
    at Function.Module._load (internal/modules/cjs/loader.js:790:14),
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:76:12),
    at internal/main/run_main_module.js:17:47 {,
  errno: -2,,
  syscall: 'stat',,
  code: 'ENOENT',,
  path: '/app/data/ca.srl',
},

chuot, 2021-07-09:
I reproduced it, working on a fix. Thanks for the report.

atlex, 2021-07-09:
Thank you for taking core of it.

There is no issue with: 
$ docker-compose -p rdio-scanner run -d --service-ports server

chuot, 2021-07-09:
The thing is that the very first time you are running the docker container, it will create for you the missing ~/.rdio-scanner folder as per the --volume option. Since the docker daemon is running as root, the ~/.rdio-scanner folder is created and owned by the root user, not by the one you want to use through the --user option. The solution is to change the owner for the folder ~/.rdio-scanner, or just `rm -fr ~/.rdio-scanner` and recreate it. Then restart the docker container.

I will adjust the documentation accordingly.

Please close this issue if it's resolved for you.

Thanks

atlex, 2021-07-12:
@chuot thank you for information. It helped a lot. Now it forks fine. I created ~/.rdio-scanner with normal user(not root). My steps:

$ mkdir ~/.rdio-scanner
$ docker pull chuot/rdio-scanner:latest
$ docker run --detach --env TZ=Europe/Minsk --name rdio-scanner --publish 3000:3000 --restart always --user $(id -u):$(id -g) --volume ~/.rdio-scanner:/app/data --volume ~/recs:/home/radio/recordings chuot/rdio-scanner:latest

closed 2021-07-12 by atlex