chuot/rdio-scanner #48: Feature request. Add more file name formats for Dirwatch.

atlex, 2021-07-12:
Hi @chuot,

Could you please parse files like 2021_07_12_17-34-07.wav (YYYY_MM_DD_hh-mm-ss)? It's output of popular SDRSharp https://airspy.com/download/.
Or provide more variables in the mask like #SDATE and #STIME. 

There is no other way in the SDRSharp to use other date time format :(

chuot, 2021-07-13:
This is a good idea. Would it be better to have a specific dirwatch type like for trunk-recorder and sdr-trunk? I can also modify the current #DATE and #TIME to take into account the underscores for the date and the dashes for the time so #DATE and #TIME.

I guess that a new sdr-sharp type would be better if the wav files also come with metadata from another (json) file. If not, then just changing the regexp for #DATE and #TIME.

I will look at it and should come with something for the next release.

chuot, 2021-07-20:
Fixed in v5.2.1

closed 2021-07-20 by chuot