chuot/rdio-scanner #52: m4a fails to play

ZeroChaos-, 2021-08-28:
I am using the docker container from yesterday.  When I monitor for mp3 files everything works fine, but when I monitor for m4a files the screen lights up (like it's going to play) but no audio is produced).  I am encoding mp3 and m4a, and ffplay/vlc/etc can play either file fine, but for some reason the m4a's don't make audio in rdio-scanner.

Any thoughts?

chuot, 2021-08-30:
1. Please describe your environment, browser + OS
2. Can you provide any error message from the javascript console of your browser
3. Can you provide an example of a m4a that doesn't play. Please extract it directly from the Rdio Scanner webapp (search panel).

ZeroChaos-, 2021-08-31:
so the actual bug was that ffplay/ffmpeg is WAAAAY too forgiving on making/playing an aac file and mine was invalid.  Now that I am encoding it correctly there is no issue.  My apologies for the noise.

closed 2021-08-31 by ZeroChaos-