chuot/rdio-scanner #53: Feature Request

blackwellj, 2021-08-28:
Would it be possible to have a play unheard option? This would be nice to listen back to a conversation

chuot, 2021-08-30:
Keeping track of all the already listened calls as to be done directly on the webapp, because it depends of what the user is listening to. This could end up in a huge amount of callIds stored locally. I my case, I just filter calls from the search panel, then playback them back in order while the live feed button is off. This put Rdio Scanner in playback mode were the calls are played automaticaly one after the other, following the filtered list from the search panel. If I've already heard a call, then I simply click the skip next button.

blackwellj, 2021-08-30:
OK, I don’t think the automatic playback feature is working on the ios app. Could a variable be added that would log the last listened time? that would only be a few bytes?

chuot, 2021-08-30:
Yes it does on both native apps, and the webapp. Just make sure that the Live Feed is off (red) prior to clicking on a play button from the search panel. Then at the end of the call, it will load the next one and so on. If you go back to the main screen, you'll see the Live Feed button is lit in orange.

blackwellj, 2021-08-30:
ok, ill give that a try

closed 2021-09-02 by blackwellj