chuot/rdio-scanner #59: Node Update Fails

M4ttjabz, 2021-09-27:
Trying to update and it keeps failing with this error:

  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/home/ubuntu/rdio-scanner/server/node_modules/bcrypt/bcrypt.js' ]

now I can't run the server.

Any ideas?

Thanks!

Matt

chuot, 2021-09-27:
Which nodejs version you're on? Assuming it's nodejs v14 and up, try "npm ci" in the server folder then retry the update.

M4ttjabz, 2021-09-28:
Thanks Chrystian!

I tried a few times with no luck, then waited a couple hours and it worked. (maybe some kind of network issue?) Anyway, it finally built successfully.  I was hoping this update would allow me to use the iPhone app but I'm still not getting any calls on the app.  I've allowed port forwarding and have a static IP; so the web-interface works anywhere but no luck with the app despite it saying "Connected" in green.   Any other tips?

chuot, 2021-09-28:
Rdio Scanner deals with audio files recorded by various type of recorders, which aren't part of Rdio Scanner. Which recorder are you using and how did you configured Rdio Scanner to ingest them, a dirWatch, an upload script?

M4ttjabz, 2021-09-28:
I'm using Trunk Recorder and dirwatch.  It works great but the iPhonw app still doesn't work despite saying connected.  Also, after about a day of recordings (on a very busy smartnet system); Rdio player crashes until I start deleting recordings.  I've got 80gb of free space but it seeems there's a limit to how many files dirwatch will allow.  As soon as I delete a day's worth of recordings; it works perfectly.

chuot, 2021-09-29:
Please send me by email the url to your instance. I will test and try to understand what's happening with it.

Also, please consider to close this issue since it's no longer related the its subjet. The Gitter forum is a better place for such kind of troubleshooting.

Thanks

chuot, 2021-11-16:
No updates, closing.

closed 2021-11-16 by chuot