chuot/rdio-scanner #65: Add support for talkgroup patching info currently being added to trunk-recorder

aaknitt, 2021-11-30:
See [here](https://github.com/robotastic/trunk-recorder/issues/368) and [here](https://github.com/robotastic/trunk-recorder/pull/562/) for background.

Currently working on adding patch-following to trunk-recorder, at least for Motorola-style patching.  The way Motorola patching works, the subscriber radios track the patches.  When a transmission occurs it is only sent on one of the patched talkgroups (the "supergroup" TGID).  Anyone monitoring one of the other patched TGIDs with trunk-recorder/rdio-scanner would be unaware of the traffic.

Current plan is to have trunk-recorder track the patches like a subscriber radio would and add a list of all patched talkgroups to the JSON file for the "supergroup" TGID that actually gets recorded, and to also somehow make the patching information available via the plugin system (mechanics of that are still TBD).

As we're working on this I'd be interested in your input so that we can make sure that this new data can be utilized by rdio-scanner.

Example scenario:
System has talkgroups A, B, C, D, and E on it.  rdio-scanner user has selected to listen to C and D.  A patch of talkgroups A, B, and C is created, with A being the supergroup that gets sent over the air.

Ideally, rdio-scanner would recognize that even though the transmission is going out on A and so it would normally ignore it, it can see that C (which the user does want to listen to) is part of the patch, and so would play the audio file for A with indication on the GUI that the A-B-C patch exists.

Let me know either here or on the trunk-recorder issue thread how we can set up the trunk-recorder code/JSON/plugin data to best enable this type of feature.

chuot, 2021-11-30:
That would be great!

Just a new field named `patches` with a list of patched TGIDs, and the SUPERTGID in the `talkgroup` field for both the plugin and the json file.

It won't make it for v6.0, but I will sure add this feature for v6.1. The search panel will require more work as it will need changes to the database structure in regards to archived calls.

aaknitt, 2021-12-06:
Code under development is here until it gets merged to the main trunk-recorder branch:
https://github.com/aaknitt/trunk-recorder

Interfaces are as follows:
JSON data now includes an optional field called patched_talkgroups, only present if the talkgroup is part of a patch:
```
"talkgroup": 58918,
"talkgroup_tag": "WKLEDISP5",
"patched_talkgroups": [58916,58918],
```
In the plugin, a list of patched talkgroups associated with a given call can be retrieved in call_start() as follows (or anywhere else that a Call object is available):
```
long talkgroup_num = call->get_talkgroup();
std::vector<long> patched_talkgroups = call->get_system()->get_talkgroup_patch(talkgroup_num);
```
and in call_end() as follows:
```
long talkgroup_num = call_info.talkgroup;
std::vector<long> patched_talkgroups = call_info.patched_talkgroups;
```

chuot, 2021-12-06:
That's great! I need to stabilize the app in its current state and track down those SQL_BUSY errors, but then I'll add this new functionality to Rdio Scanner. Like I said, this will bring a minor change in the structure of the database, but the price is minimal compared to what Rdio Scanner will earn by listening to patched talkgroups and searching the archives.

aaknitt, 2021-12-12:
This has been merged into the trunk-recorder master branch, just FYI.

chuot, 2021-12-13:
Thanks, I'm getting to it .... soon

chuot, 2022-01-04:
Not tested much, but it should be ok.

Note1: when in LIVEFEED, calls from avoided talkgroups will still play if there are in a supergroup that it isn't avoided.

Note2: it is also possible to search archived calls in the same maner, but it takes much longer to search for a specific talkgroup that is included in a supergroup. Consequently, searching in patched_talkgroups is deactivated by default but can be enabled from the admin / config / options.

I'm closing this issue, but please reopen if needed.

aaknitt, 2022-01-05:
> Note1: when in LIVEFEED, calls from avoided talkgroups will still play if there are in a supergroup that it isn't avoided.

I think we actually need the opposite.  The audio recorded on the supergroup talkgroup needs to play even when it's being avoided if any of the patched talkgroups is not being avoided.

As you've described it, I'm not sure the functionality is any different from before...if a supergroup is not being avoided, it will play.  

I think maybe there's some confusion as to how trunk-recorder is handling the patching.  An audio file is only ever created for the supergroup.  The JSON file associated with the supergroup audio file will have the list of other talkgroups that are part of the patch, but no audio files (or JSON files) get created for those patched talkgroups.  

So I think rdio-scanner would need to look at every created JSON file and check to see if any of the patched talkgroups are being monitored in the livestream.  If so, play the (supergroup) audio file.  Ideally there would be some visual indication that it is being played due to patching (perhaps even listing all TGIDs in the patch?), otherwise users might get confused when an audio file from a supergroup they are avoiding is playing due to a patch.  

@chuot I'm not able to reopen the issue since I didn't close it, but would suggest doing so.  In the small amount of testing I've done the functionality doesn't seem to be any different from in the past.

chuot, 2022-01-05:
I think it is working as intended. Let's say that your Rdio Scanner client is set like so:

- tg 58916 is enabled (not avoided)
- tg 58918 is disabled (avoided)

Then a new call arrives from trunk-recorder

```json
"talkgroup": 58918,
"patched_talkgroups": [58916,58918]
```

The call will play on the client only if

- tg 58918 is enabled, or
- *one of the tgs* in the `patched_talkgroups` is enabled.

In this scenario, 58916 is enabled, so the call will be sent to the client.

The one thing I wasn't aware of is that trunk-recorder won't create audio files for patched talkgroups, only for the supergroup. This means that a dirwatch with deleteAfter set to true won't delete the json file if there's no audio associated with it. I will have to change that.

As for call the plugin, I guess that it won't be called if there's no audio to send, but just once for the supergroup. This has to be confirmed.

aaknitt, 2022-01-05:
I'm using the trunk-recorder plugin with the API.  Looking at the plugin code, it looks like the plugin isn't actually uploading the JSON file, but is instead sending data passed in via call_info.  In that case, I think an update to the plugin is needed in order to support this functionality (retrieve the patching info from call_info and send it to the API)?  Does an update exist somewhere that I can grab it and try it?

aaknitt, 2022-01-05:
If you define the API interface to send the data over I can try to help with the C++ stuff.  I'm not an expert either, but have been doing some work in trunk-recorder lately.

chuot, 2022-01-05:
It is already documented in the `docs/api.md` document.

Basically, you just have to pass the patched talkgroups in json using the field name `patches`, as per the document.

chuot, 2022-01-05:
The doc needs some updating too, see issue #68

aaknitt, 2022-01-06:
Got it, I think I have this working now.  I'll submit a PR to trunk-recorder with the update.

aaknitt, 2022-01-06:
https://github.com/robotastic/trunk-recorder/pull/597

chuot, 2022-01-06:
I just tested it and it appears to be working fine.

Thanks a lot!

closed 2022-01-07 by aaknitt