chuot/rdio-scanner #66: read talkgroup_tag and src tag from json

ZeroChaos-, 2021-12-01:
Trunk-recorder supports tagging for both UID/src as well as talkgroup.  This information is already available in the json.  Please read and display this available info
```
{
    "freq": 460050000,
    "start_time": 1638371580,
    "stop_time": 1638371602,
    "emergency": 0,
    "encrypted": 0,
    "call_length": 20,
    "talkgroup": 1,
    "talkgroup_tag": "Moon Police",
    "freqList": [
        {
            "freq": 460050000,
            "time": 1638371580,
            "pos": 0.0,
            "len": 20,
            "error_count": 0.0,
            "spike_count": 0.0
        }
    ],
    "srcList": [
        {
            "src": 1545,
            "time": 1638371580,
            "pos": 0,
            "emergency": 0,
            "signal_system": "",
            "tag": "Moon PD 609"
        }
    ]
}
```

chuot, 2021-12-01:
Should work with v6.0.1-beta.

closed 2021-12-01 by chuot

ZeroChaos-, 2021-12-01:
awesome, thanks!  Any chance there is a nice convenient docker container for testing that?  :-)