chuot/rdio-scanner #67: V6.0.1 - SQL Lock Errors

JayManCanada6014, 2021-12-01:
Posted in the chat as well.

Ubuntu 20.04 using the Linux AMD64 6.0.1 file. When I first start the app It starts to ingest the calls and auto populate. The moment I uploaded my config from v5 I got these.  And each time it tried to write a call to the DB after the upload it got the lock messages. If I restart Rdio-Scanner it goes back to the default settings and starts to auto pop the TGs and Systems.

root@rdio-scanner:/opt/rdio-scanner-beta# ./rdio-scanner

Rdio Scanner v6.0.1
2021/12/01 15:04:07 dirwatch.isDir: open /scan-watch: no such file or directory
2021/12/01 15:04:07 dirwatches.start: stat /scan-watch/.: no such file or directory
2021/12/01 15:04:07 dirwatch.isDir: open /watch/EASalerts: no such file or directory
2021/12/01 15:04:07 dirwatches.start: stat /watch/EASalerts/.: no such file or directory
2021/12/01 15:04:07 server started
2021/12/01 15:04:07 main interface at http://rdio-scanner:3001
2021/12/01 15:04:07 admin interface at http://rdio-scanner:3001/admin
2021/12/01 15:04:12 newcall: system=104 talkgroup=38096 file=38096-1638389032_142995000-call_15157.m4a success
2021/12/01 15:04:13 newcall: system=104 talkgroup=38096 file=38096-1638389032_142125000-call_146.wav duplicate call rejected
2021/12/01 15:04:17 newcall: system=104 talkgroup=38096 file=38096-1638389048_142005000-call_15158.m4a success
2021/12/01 15:04:18 newcall: system=104 talkgroup=38096 file=38096-1638389048_141675000-call_147.wav duplicate call rejected
2021/12/01 15:04:18 newcall: system=100 talkgroup=8960 file=8960-1638389032_141435000-call_145.m4a success
2021/12/01 15:04:29 newcall: system=104 talkgroup=38096 file=38096-1638389051_141375000-call_15159.m4a success
2021/12/01 15:04:30 controller.ingestcall: systems.write: database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:30 admin.confighandler.put: groups.write database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:30 admin.confighandler.put: options.write: database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:30 admin.confighandler.put: systems.write: database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:30 admin.confighandler.put: tags write database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:30 dirwatch.isDir: open /scan-watch: no such file or directory
2021/12/01 15:04:30 dirwatches.start: stat /scan-watch/.: no such file or directory
2021/12/01 15:04:30 dirwatch.isDir: open /watch/EASalerts: no such file or directory
2021/12/01 15:04:30 dirwatches.start: stat /watch/EASalerts/.: no such file or directory
2021/12/01 15:04:30 configuration changed
2021/12/01 15:04:35 controller.ingestcall: systems.write: database is locked (5) (SQLITE_BUSY)
2021/12/01 15:04:35 controller.ingestcall: systems.write: database is locked (5) (SQLITE_BUSY)

JayManCanada6014, 2021-12-01:
Also even tried to set the directory to 777 just to make sure and that didn't help.

chuot, 2021-12-01:
No, it's something with the SQL driver. Investigating...

chuot, 2021-12-02:
Fixed in v6.0.2-beta

closed 2021-12-02 by chuot