chuot/rdio-scanner #68: (error)   Plugin Manager: call_end -  rdioscanner_uploader failed

M4ttjabz, 2021-12-04:
Hi Christian- 

Please forgive me if this has already been addressed:

I'm using the latest OP25 build of TrunkRecorder with the new Rdio-Scanner plug-in.  

I've upgraded to 6.0 v2 - its awesome that its its a binary and doesn't need to be compiled!!!

My only issue is with the new plug-in for Trunk-recorder:
I can't seem to get the plug-in to talk to Trunk-recorder.   I keep getting:  "  (error)   Plugin Manager: call_end -  rdioscanner_uploader failed"  

My trunk recorder config is: 

   }],
         "plugins": [{
          				"name": "rdioscanner_uploader",
         				"library": "librdioscanner_uploader.so",
          				"server": "http://127.0.0.1",
          				"systems": [{
                  			"shortName": "CHO",
                  			"apiKey": "d72ea39a-ad26-44c6-a1c7-49d6c10d1e6e",
                  			"systemId": "15"
                     }]
        }],
        "defaultMode": "digital",
        "captureDir": "/home/matt/trunk-build/calls",
        "callTimeout": 4,
        "logFile": true,
        "logLevel": "debug",
        "frequencyFormat": "mhz"
}

API, shortname and systemId all match up.

Thanks!

-Matt

M4ttjabz, 2021-12-04:
I think I figured it out:
For anyone with the same issue:
127.0.0.1 doesn't work ---  you need to use the PC name, ie. ""server": "http://trunkrf:3000","  .   
This may need to be changed on the Trunk-Recorder ReadMe - unless I'm the only one who didn't know that?   which is very possible...

Anyways-  Thank you Christian for all your hard work on this!!!

aaknitt, 2021-12-05:
Ran into the same issue and had the same resolution.  The docs do say to use the same URL as is used to access the GUI, but pointing out that the port is needed would be helpful.

chuot, 2021-12-05:
Will look at it the week

aaknitt, 2021-12-07:
Actually I just re-read this and realized that my resolution was different than what the OP noted.  Using 127.0.0.1:3000 worked for me.  I did not need to use the PC/host name, just needed to add the port after the PI address.

closed 2022-01-31 by chuot