chuot/rdio-scanner #69: Possible issue with trunk-recorder plugin

aaknitt, 2021-12-05:
When I was running an earlier version or rdio-scanner, I used a trunk-recorder uploadScript to copy audio files to a directory dedicated to a rdio-scanner dirwatch.  This allowed rdio-scanner to delete the files to keep good performance while still allowing me to archive files in the original trunk-recorder recording directory.

With this setup, I found that trunk-recorder would sometimes call the uploadScript before the audio file was actually done writing to the file system, resulting in a failed copy to the rdio-scanner dirwatch directory.  I had to add a delay of 100 to 200 ms into my uploadScript to ensure that the audio file was fully written before attempting to copy it.

I think I may be seeing similar behavior with the rdio-scanner plugin for trunk-recorder.  When using the plugin, I will sometimes see calls on the GUI but no audio will play either from the live feed or using the "search call" function.  I suspect that this may be for similar reasons - trunk-recorder may be calling the call_end() function in the plugin before the audio file is fully written. 

I'm not certain this is what's actually happening.  I may try to dig into the plugin code a bit and add some debugging printouts to try to pinpoint what's going on.  If you have any suggestions for doing so let me know.

aaknitt, 2021-12-05:
After doing some more testing I'm not sure my theory holds water.  I'm seeing similar issues with a dirwatch using my old method, and I've confirmed that the audio files making it into the directory are good...they play just find outside of rdio-scanner but don't seem to be getting ingested properly for some reason.  So I think there may still be an issue, but perhaps not just with the trunk-recorder plugin?

chuot, 2021-12-05:
It think it may be because of the audio conversion done by ffmpeg.

Is Trunk Recorder has its compresswav option enabled or does it provide wav files to Rdio Scanner?

Is Rdio Scanner doing audio conversion? That conversion occurs only if ffmpeg is available to Rdio Scanner and if the disable audio conversion option is not activated.

aaknitt, 2021-12-06:
I have trunk-recorder doing the file conversion to m4a, which I need for upload to OpenMHz.  Are you saying that having rdio-scanner do that conversion instead may help with this?

chuot, 2021-12-06:
No, I'm saying that maybe it is the ffmpeg conversion that Rdio Scanner is doing if the disableAudioConversion isn't set. Since you are already converting to m4a on the Trunk Recorder side, I would recommend to try to activate the disableAudioConversion from the admin dashboard. If you still noticing missing audio from Rdio Scanner, then the problem isn't with ffmpeg. But first, we need to exclude that one from the problem.

aaknitt, 2021-12-07:
That seems to have done the trick.  No more issues noted with either the plugin/API or dirwatch after disabling audio conversion within rdio-scanner.  Thanks!

closed 2021-12-07 by aaknitt