chuot/rdio-scanner #70: rdio-scanner failed to write rdio-scanner.db error unhelpful

ZeroChaos-, 2021-12-05:
I realize that '-db_file' config param can be given to move the db file, so this isn't a support request.  That said, when rdio-scanner is installed in /usr/bin (for example) and a user attempts to run it, a very unhelpful error pops up
```
2021/12/05 14:47:31 unable to open database file: out of memory (14)
```
Would it be possible to detect that it's a file permission error and provide a more useful error suggesting to use -db_file to put the db somewhere more sane?

chuot, 2021-12-06:
The SQL sub-package returns this error which I agree is not meaningful. I may add a test for this or I may rethink the way Rdio Scanner looks for its database and its config file. One of the goal here is to keep the experience harmonized across all platforms. I wouldn't recommend to put Rdio Scanner in /usr/bin, but rather in /usr/local/rdio-scanner, /opt/rdio-scanner or $HOME/rdio-scanner.

ZeroChaos-, 2021-12-06:
Following the filesystem hierarchy standard if I install it from the package manager it does in /usr/bin, although since I'm shipping the binary version right now I'm putting it in /opt/rdio-scanner-bin (the -bin bit is a gentoo convention).  I've packaged rdio scanner in the pentoo overlay for now because I really don't like shipping binaries but I also can't get the latest version to build right in the package manager :-)

anyway, if you find a sane way to pop a more useful error that would be great.  probably easiest to just add a write check for the location before you try to open/create the file.

chuot, 2021-12-08:
As of version 6.0.4, if Rdio Scanner cannot write files to the same folder as the executable, it will create an Rdio Scanner folder in the user's home folder where data will be written.

If no folder is available for writing, a more meaningful error message is displayed and the application closes.

I also added a COMPILING.md file to help you out.

Do not hesitate to reopen if necessary.

closed 2021-12-08 by chuot