chuot/rdio-scanner #71: Losing Imported Talkgroup Info After Reboot on RasPi with v6.0.5

aaknitt, 2021-12-11:
Using the 6.0.5 Arm binary on a Raspberry Pi.  After using the CSV import feature to get my talkgroup definitions loaded in, all works well until after I reboot the Pi.  At that point only a small handfull (10 out of ~100) of talkgroups populate in the "Select TG" screen.

aaknitt, 2021-12-11:
The talkgroups that remain after a reboot _may_ be ones that were recorded prior to the reboot, but I'm not certain of that correlation yet.

chuot, 2021-12-13:
Thanks for reporting the issue. I will try to reproduce on a RPI.

chuot, 2021-12-13:
Please confirm that you are running Rdio Scanner as a service and have run a shutdown command on the PI? I think I identified the problem when shutting down the PI (or any other platform), Rdio Scanner was exiting without properly closing the database. Because a PI is slower than any other platform, I guess the problem has become more obvious.

chuot, 2021-12-13:
I'm hoping v6.0.6 will resolve this issue. Please feel free to reopen if it isn't the case. Thanks

closed 2021-12-13 by chuot

aaknitt, 2021-12-14:
Confirmed, seems to be fixed, thanks!