chuot/rdio-scanner #72: talkgroup storage in v6-beta loses talkgroups

gisforgirard, 2021-12-15:
talkgroups really should be in their own table, not stored in a json object inside a single table row... try importing this system and watch the havoc that occurs

[trs_tg_100.csv](https://github.com/chuot/rdio-scanner/files/7722904/trs_tg_100.csv)

gisforgirard, 2021-12-15:
nevermind i have no idea what happened but wiping the entire tatabase and starting over at least made things work again

gisforgirard, 2021-12-16:
SqlError: (conn=87, no: 1406, SQLState: 22001) Data too long for column 'talkgroups' at row 1
actually this explains a lot

chuot, 2021-12-17:
I agree with you, talkgroups should reside in their own table. I will work that out for v6.1 which will come out somewhere in january. Thanks for reporting the issue.

closed 2021-12-23 by gisforgirard