chuot/rdio-scanner #73: uid -1 issue

ZeroChaos-, 2021-12-20:
Hi again.  I opened an issue over on trunk-recorder a while ago because the behavior of TR changed between 3.x and 4.x with respect to unidentified transmissions (I'm assuming analog only, with no mdc1200 or similar).

https://github.com/robotastic/trunk-recorder/issues/541

I'll let you read the TR bug, but long and the short of it is that TR now passes "-1" fo the uid in cases of unidentified transmissions to explicitly say "I got no clue" using a uid which isn't valid for any system we are aware of.  This works fine in rdio-scanner 5.x, but apparently not in 6.x

I hope the fix isn't a big deal for you, but if it is, please feel free to discuss here, I'll cross link the bugs.

taclane, 2021-12-20:
In the linked thread, it was mentioned that this should be addressed in 6.1.  

If others are noting this issue, -1 UIDs sent out via trunk-recorder get stored by rdio-scanner 6.x as `18446744073709551615`. The web interface will display these preposterously large UIDs, and the mobile clients seem unable to load those transmissions.

chuot, 2021-12-20:
My bad... Due to the way Go works, I had to normalize all numerical data types to something common. I chose ussigned int because it made sense at the time, but now that you're having this problem, I think I chose the wrong type and it would have been better to choose int instead. I will go through all the code in the upcoming v6.1 and will change uint to int. Therefore, you should see uid: -1 instead of the current ugly number.

chuot, 2022-01-04:
Fixed in v6.1

closed 2022-01-04 by chuot

ZeroChaos-, 2022-01-04:
woohoo!  thanks!