chuot/rdio-scanner #74: v6.0.7-beta Can't change configuration SQLITE_BUSY Ubuntu

atlex, 2021-12-23:
rdio-scanner-linux-amd64-v6.0.7.zip
Ubuntu 20.04.3 LTS x86_64

$ ./rdio-scanner 

http://host1:3000/admin
Config > Tags
Remove some default tag.

Logs:

2021/12/23 12:33:16 server started
2021/12/23 12:33:16 base folder is /home/pi/.rdio-scanner
2021/12/23 12:33:16 main interface at http://host1:3000
2021/12/23 12:33:16 admin interface at http://host1:3000/admin
2021/12/23 12:33:36 admin.confighandler.put: tags write database is locked (5) (SQLITE_BUSY)
2021/12/23 12:33:36 configuration changed

The same for Config > Systems.

chuot, 2021-12-23:
Thanks for reporting the issue.

taclane, 2021-12-27:
I've been having the issue on 6.0.7 as well.  Sometimes config changes go through, sometimes they don't.

I've even tried the usual trick of making a backup of the database with
`Sqlite> .backup main backup.sqlite`
to remove any unresolved locks, and the issues persist with the backup copy.

chuot, 2022-01-04:
It should be ok in v6.1

closed 2022-01-04 by chuot