chuot/rdio-scanner #78: Rdio-scanner 6.x using .wav extension for .mp4 files

taclane, 2022-01-05:
Something I've noticed in the transition to Rdio-scanner 6 (and the trunk-recorder plugin) is that mp4 files are being served up with a .wav file extension.

Trunk-recorder is compressing locally, and as far as the transfer logs show, appears to be sending a file approximately the right size for mp4 audio.  For instance, this is a ~40 second call:
```
Rdio Scanner Upload Success - file size: 231379
```
Rdio-scanner is set to disable conversion, and on its end, the logs show something like this when each new call is uploaded:
```
newcall: system=4545 talkgroup=2 file=2-1641417524_455375000-call_1.wav success
```
In the "Search Call" interface, individual calls can be downloaded, and they will be saved with a .wav file extension.

Audio players will recognize those .wav files as mp4.  Manually inspecting the file shows that it has a `ftypM4A` header.

taclane, 2022-01-06:
Seems like the problem is strictly on the plugin side.  Unlike other values that change whether or not `compress_wav` is true, `audioName` does not, and it always reports back the .wav filename.  Just to check, I manually edited line 85 of the plugin to use 
```
boost::filesystem::path audioPath(call_info.converted);
```
and things are fine; `audioName` just needs to test if it has a m4a file before sending off the incorrect filename.

https://github.com/robotastic/trunk-recorder/blob/05f0a12fa6c51056235fc5b809362e61b51f30dd/plugins/rdioscanner_uploader/rdioscanner_uploader.cc#L131-L135

chuot, 2022-01-06:
Indeed, it appears that the plugin is sending different informations...

When compressWav is false, the audio is sent as octet-stream with filename.m4a, the audioName is filename.wav and the audioType is audio/mp4.

I'm going to tighten things up at the API level, but this needs to be fixed at the plugin level.

Thanks for reporting this issue.

taclane, 2022-01-24:
The above PR should address the plugin side of things once it get merged in.

taclane, 2022-01-26:
The PR was merged on trunk-recorder's end.  The plugin should now post the correct filename when uploading to rdio-scanner.

closed 2022-01-26 by taclane