chuot/rdio-scanner #85: Android App keeps stopping

ejsilver26, 2022-01-09:
### Discussed in https://github.com/chuot/rdio-scanner/discussions/84

<div type='discussions-op-text'>

<sup>Originally posted by **ejsilver26** January  9, 2022</sup>
I am using the android app for rdio. The app keeps stopping at random times. I have searched through all the setting that I can find to keep it running, even allowing it to run when the "Do Not Disturb" is on. I have a Samsung phone, run android 11, and am running rdio 1.3.2. Thanks. </div>

I suppose this is an issue as well.

chuot, 2022-01-10:
On your Samsung device, go to Settings / Apps, click on the three little dots on the upper right corner and select Special Access. Then select Optimize Battery Usage, change the filter from Apps Not Optimized to All and search for Rdio Scanner in the list. The toggle button on the right side to Rdio Scanner should (must) be disabled. This will tell Android not to kill the app to spare battery life.

When you have installed the app, the first time you have launced it, while swiping through the welcome slides, the one before the last one tells about that step.

If it still being killed, then maybe you are using some sort of a battery optimizer app that kills Rdio Scanner. Then you will have to configure that battery optimizer to let Rdio Scanner alone.

ejsilver26, 2022-01-10:
It was disabled. I have not installed any other battery optimizer, nor do I have another as far as I know. Sometimes it will run for hours, sometimes not. I wonder if it is related to Wifi? We live in the country and sometimes the connection will get dropped for a minute or two. Usually doesn't affect anything else (other than buffering). I noticed that the app turns off when I get home and it switches from Data to Wifi, and vice versa. Is there a way to verify that?

chuot, 2022-01-18:
I will harden the way the native app reconnects to the server when switching networks.

ejsilver26, 2022-01-18:
> 

That will probably help. If you need a Guinee pig, let me know. :)

chuot, 2022-01-31:
Did the latest version resolved your issue?

ejsilver26, 2022-02-02:
I have not noticed any difference, but I am also stuck at home.
------------------------

Eli Silverman

[image: Eli Silverman on about.me]