chuot/rdio-scanner #86: rdio-scanner cannot create tables in an empty MySQL database

VVakko, 2022-01-09:
It was a challenge to switch rdio-scanner from SQLite to MySQL database. :)

Background.
Before the New Year holidays, I once again set up rdio-scanner and trunk-recorder for 3 systems. During the week of the New Year holidays, the sqlite database grew to 2.3GB. Next, I decided to update rdio-scanner from version 5.2.9 to version 6.1.1. I updated it and everything seemed to work properly for a while.
But today I saw that the rdio-scanner utilizes 100% of the CPU. This container is executed on Synology DS 1618+ with a simple Atom C3538 processor. I turned off the trunk-recorder containers completely and waited for about an hour. The processor was still loaded. I didn't understand what rdio-scanner was so busy with, perhaps deleting old records (pruneDays = 7).

In general, I decided to try switching the database to an external MySQL database. Created a new database and user, created a `rdio-scanner.ini` file with the following content:
```ini
db_type = mysql
db_host = server
db_port = 3306
db_name = db_name
db_user = db_user
db_pass = db_pass
listen = :3000
```

I run rdio-scanner and got an error:
```
2022/01/09 20:31:16 running database migration 20191028144433-create-rdio-scanner-system
2022/01/09 20:31:16 Error 1101: BLOB, TEXT, GEOMETRY or JSON column 'talkgroups' can't have a default value while doing create table `rdioScannerSystems` (`id` integer primary key auto_increment, `createdAt` datetime not null, `updatedAt` datetime not null, `name` varchar(255) not null, `system` integer not null, `talkgroups` json not null default '[]')
```

Next, I took an empty `rdio-scanner.db`, exported the schema to a .sql file and tried to import it to the MySQL database. It didn't work out directly, I had to edit the file a little manually. Keyword `autoincrement` changed to `auto_increment` and near the fields `blacklists`, `frequencies`, `patches` and `sources` I have to delete ` default '[]'`, because in MySQL columns of the type `BLOB`, `TEXT`, `GEOMETRY` or `JSON` can't have a default value.
The result is this .sql file:
```sql
CREATE TABLE `temp_rdioScannerAccesses` (`_id` integer primary key auto_increment, `code` varchar(255) not null unique, `expiration` datetime, `ident` varchar(255), `limit` integer, `order` integer, `systems` text not null);
CREATE TABLE `rdioScannerApiKeys` (`_id` integer primary key auto_increment, `disabled` tinyint(1) default 0, `ident` varchar(255), `key` varchar(255) not null unique, `order` integer, `systems` text not null);
CREATE TABLE `rdioScannerCalls` (`id` integer primary key auto_increment, `audio` longblob not null, `audioName` varchar(255), `audioType` varchar(255), `dateTime` datetime not null, `frequencies` text not null, `frequency` integer, `patches` text not null, `source` integer, `sources` text not null, `system` integer not null, `talkgroup` integer not null);
CREATE TABLE `rdioScannerConfigs` (`_id` integer primary key auto_increment, `key` varchar(255) not null unique, `val` text not null);
CREATE TABLE `rdioScannerDirWatches` (`_id` integer primary key auto_increment, `delay` integer default 0, `deleteAfter` tinyint(1) default 0, `directory` varchar(255) not null unique, `disabled` tinyint(1) default 0, `extension` varchar(255), `frequency` integer, `mask` varchar(255), `order` integer, `systemId` integer, `talkgroupId` integer, `type` varchar(255), `usePolling` tinyint(1) default 0);
CREATE TABLE `rdioScannerDownstreams` (`_id` integer primary key auto_increment, `apiKey` varchar(255) not null, `disabled` tinyint(1) default 0, `order` integer, `systems` text not null, `url` varchar(255) not null);
CREATE TABLE `rdioScannerGroups` (`_id` integer primary key auto_increment, `label` varchar(255) not null);
CREATE TABLE `rdioScannerLogs` (`_id` integer primary key auto_increment, `dateTime` datetime not null, `level` varchar(255) not null, `message` varchar(255) not null);
CREATE TABLE `rdioScannerMeta` (name varchar(255) not null unique primary key);
CREATE TABLE `rdioScannerSystems` (`_id` integer primary key auto_increment, `autoPopulate` tinyint(1) default 0, `blacklists` text not null, `id` integer not null unique, `label` varchar(255) not null, `led` varchar(255), `order` integer);
CREATE TABLE `rdioScannerTags` (`_id` integer primary key auto_increment, `label` varchar(255) not null);
CREATE TABLE `rdioScannerTalkgroups` (`_id` integer primary key auto_increment, `frequency` integer, `groupId` integer not null, `id` integer not null, `label` varchar(255) not null, `led` varchar(255), `name` varchar(255) not null, `order` integer, `systemId` integer not null, `tagId` integer not null);
CREATE TABLE `rdioScannerUnits` (`_id` integer primary key auto_increment, `id` integer not null, `label` varchar(255) not null, `order` integer, `systemId` integer not null);

INSERT INTO `rdioScannerMeta` (`name`) VALUES
('20191028144433-create-rdio-scanner-system'),
('20191029092201-create-rdio-scanner-call'),
('20191126135515-optimize-rdio-scanner-calls'),
('20191220093214-new-v3-tables'),
('20200123094105-optimize-rdio-scanner-calls'),
('20200428132918-new-v4-tables'),
('20210115105958-new-v5.1-tables'),
('20210830092027-v6.0-rename-index'),
('20211202094819-v6.0.2-alter-table'),
('20220101070000-v6.1.0');
```

Then there was another blunt with importing the configuration from a .json-file. It was not quite obvious that after the import, it was necessary to click on the Save button, otherwise the import was not performed.

In general, everything is working now, the CPU load is near zero.

chuot, 2022-01-10:
MariaDB isn't exactly like mySQL in regards to that. Unfortunately, I'm using the same SQL driver for both (the only one that compiles on various platforms). So I have to have two sets of queries, one for mariadb the the other one for mariadb/mysql. Those for mariadb/mysql have been tweaked so they won't have the default value anymore. It will be something to test again when v6.1.3 will come out.

As for importing the config, indeed, you have to save after. This is because sometimes the imported configuration doesn't validate and needs to be reviewed prior to saving.

VVakko, 2022-01-15:
For some reason, I decided to upgrade the container with `rdio-scanner` to version 6.1.3. This decision was totally wrong. :)
Version 6.1.3 does not work with the MySQL database at all. Version 6.1.2 just couldn't create tables in an empty database. Version 6.1.3 cannot create tables in an empty database:
```python
2022/01/15 14:31:36 Error 1101: BLOB, TEXT, GEOMETRY or JSON column 'frequencies' can't have a default value while doing create table `rdioScannerCalls2` (`id` integer primary key auto_increment, `audio` longblob not null, `audioName` varchar(255), `audioType` varchar(255), `dateTime` datetime not null, `frequencies` text not null default '[]', `frequency` integer, `source` integer, `sources` text not null default '[]', `system` integer not null, `talkgroup` integer not null)
2022/01/15 14:31:39 running database migration 20210115105958-new-v5.1-tables
2022/01/15 14:31:39 Error 1050: Table 'rdioScannerAccesses' already exists while doing create table `rdioScannerAccesses` (`_id` integer primary key auto_increment, `code` varchar(255) not null unique, `expiration` datetime, `ident` varchar(255), `limit` integer, `order` integer, `systems` text not null)
2022/01/15 14:31:43 running database migration 20210115105958-new-v5.1-tables
2022/01/15 14:31:43 Error 1050: Table 'rdioScannerAccesses' already exists while doing create table `rdioScannerAccesses` (`_id` integer primary key auto_increment, `code` varchar(255) not null unique, `expiration` datetime, `ident` varchar(255), `limit` integer, `order` integer, `systems` text not null)
...
```

Если запустить версию 6.1.3 с имеющейся таблицей, то после первой попытки загрузки wav-файла из trunk-recorder – rdio-scanner начинает выжирать 100% CPU и больше ничего не происходит. В логах пусто. Я пробовал обнулять таблицы `rdioScannerCalls`, `rdioScannerLogs`, `rdioScannerTalkgroups`, но всё так же ничего не работало.
If you run version 6.1.3 with the existing tables, then after the first attempt to upload a .wav file from trunk-recorder – `rdio-scanner` starts to utilize 100% of the CPU and nothing else happens. The logs are empty. I tried to reset the tables `rdioScannerCalls`, `rdioScannerLogs`, `rdioScannerTalkgroups`, but still nothing worked.

To return the container with version 6.1.2 turned out to be a real quest. Unfortunately, there is only the `rdio-scanner:latest` image in the docker hub, previous versions are not saved. And it is impossible to simply download an image with a specific version.
In order to make an image with version 6.1.2 on Ubuntu, I had to install a bunch of packages: `fuse-overlayfs git golang make npm pandoc texlive-latex-extra uidmap zip`. But then it turned out that NodeJS installed version 10, and you need at least 12. Go installed version 1.13, and you need at least 1.16, etc.
Now I'm writing a `Dockerfile` for myself so that I can build a docker image of any version with just two commands without having to install a bunch of dependencies.

VVakko, 2022-01-15:
`Dockerfile`:
```Dockerfile
#-------------------------------------------------------------------------------
# Base Part Used for Building RDIO-Scanner
#-------------------------------------------------------------------------------

FROM ubuntu:21.10 as builder

ENV DEBIAN_FRONTEND="noninteractive"

# Install Ubuntu updates
RUN apt-get -y update \
 && apt-get -y upgrade \
 && apt-get -y install \
 && apt-get clean

# Install packages and cleanup
RUN echo 'APT::Install-Recommends "0";\nAPT::Install-Suggests "0";\n' \
	>>/etc/apt/apt.conf.d/01norecommend \
 && apt-get update \
 && apt-get -y install \
	git \
	make \
	npm \
 && apt-get clean \
 && rm -rf /var/lib/apt/lists/*

#-------------------------------------------------------------------------------
# RDIO-Scanner (webapp building)
#-------------------------------------------------------------------------------

FROM builder as rdio-scanner-pre

# Git Clone RDIO-Scanner
ARG RDIO="master"
RUN git clone https://github.com/chuot/rdio-scanner.git \
 && cd rdio-scanner \
 && git checkout $RDIO \
 && make webapp

#-------------------------------------------------------------------------------
# RDIO-Scanner (server building)
#-------------------------------------------------------------------------------

FROM docker.io/alpine:latest
LABEL maintainer="Chrystian Huot <chrystian.huot@saubeo.solutions>"
WORKDIR /app
ENV DOCKER=1
COPY --from=rdio-scanner-pre rdio-scanner/server/. server/.
RUN mkdir -p /app/data \
 && apk --no-cache --no-progress --virtual .build add go \
 && cd server \
 && go mod tidy \
 && go build -o ../rdio-scanner \
 && cd .. \
 && rm -rf server /root/.cache /root/go \
 && apk del .build \
 && apk --no-cache --no-progress add ffmpeg tzdata
VOLUME [ "/app/data" ]
EXPOSE 3000
ENTRYPOINT [ "./rdio-scanner", "-base_dir", "/app/data" ]
```

`Makefile`:
```Makefile
RDIO = "v6.1.2"
TAG  = "rdio-scanner":${RDIO}
REGISTRY = "registry"

build-image:
	DOCKER_BUILDKIT=1 && docker build --tag ${TAG} --build-arg RDIO=${RDIO} --file ./Dockerfile .

build-image-and-publish: build-image
	docker tag ${TAG} ${REGISTRY}/${TAG}
	docker push ${REGISTRY}/${TAG}

cleanup-all: cleanup-images cleanup-volumes

cleanup-images:
	docker rmi `docker images --filter "dangling=true" -q 2>/dev/null` -q 2>/dev/null || true

cleanup-volumes:
	docker rm -v `docker ps --filter "status=exited" -q 2>/dev/null` -q 2>/dev/null || true
```

Just run `$ make build-image` for building image with specific `RDIO-Scanner` version.

chuot, 2022-01-17:
Please describe your environment, OS version, MySql version/platform. I will thighten things up with mysql on next v6.1.4.

As for the dockerfile, everything was already there in [Containerfile](https://github.com/chuot/rdio-scanner/blob/master/Containerfile). It requires you to build at least the linux-amd64 flavor prior to the container itself. Everything is describe in the [COMPILING](https://github.com/chuot/rdio-scanner/blob/master/COMPILING.md) document.

closed 2022-01-31 by chuot