chuot/rdio-scanner #88: Accept error: accept tcp [::]:3000: accept: too many open files

Brubeck48, 2022-01-10:
After reconfiguring db I got this error:

2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 5ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 10ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 20ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 40ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 80ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 160ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 320ms
2022/01/10 19:35:07 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 640ms
2022/01/10 19:35:08 http: Accept error: accept tcp [::]:3000: accept: too many open files; retrying in 1s

What is this?

chuot, 2022-01-10:
You are hitting the maximum open files limits on your linux box. Most probably because you are using a dirwatch without deleteAfter being enabled, and have a bunch of file in the monitored folder. I may be wrong here...

You can try to increase the limit on your linux box, try following [these instructions](https://www.cyberciti.biz/faq/linux-increase-the-maximum-number-of-open-files/)

Brubeck48, 2022-01-10:
I have macOS Monterey and I have tried:
`sudo launchctl limit maxfiles 65536 200000`
and `ulimit -n 10240`
but I still have this issue.

In previous version deleteAfter was enabled. There are no files in monitored folder.

chuot, 2022-01-10:
Ok, so no dirwatch then. How many users/listeners when that happens?

Brubeck48, 2022-01-10:
Only me. So I can't reach the web page after rebuilding db. It's possible only if I delete db and app create new one.

Brubeck48, 2022-01-13:
I checked everything again and I found 1 monitored folder I forgot and skipped from my previous experiments with dongles. There was a lot of files, no deleteAfter. So after emptying this directory everything working fine. Thank you!

closed 2022-01-15 by Brubeck48