chuot/rdio-scanner #89: Export button doesn't working

Brubeck48, 2022-01-15:
Hi!
I see on my macOS doesn't working export button in admin page. I tried in Firefox and Safari - nothing happening after pushing the button.

atlex, 2022-01-15:
Hi @Brubeck48,

I can't reproduce this issue. The Export button works as expected: i have output file rdio-scanner.json.

I use the Rdio Scanner v6.1.3 on macOS Catalina 10.15.7.
Google Chrome - 96.0.4664.93 (Official Build) (x86_64).
Safari 15.1.

Brubeck48, 2022-01-15:
My versions: Rdio Scanner v6.1.2
macOS Monterey 12.1
Safari Version 15.2

Good to know after upgrading from v5 to v6 it's became possible to use Rdio Scanner in Safari with self signed ssl certificate.

chuot, 2022-01-17:
I just tried on a Monterey 12.1 with Safari 15.2 on a v6.1.3 and it worked. Just after I had clicked on export, I've been prompted to allow Rdio Scanner to save the file locally, to which I answered yes.

First, you should try on v6.1.3 to see if you still have the problem. Also try running Rdio Scanner in a terminal, not in a service, so you can catch any error log.

Make sure also that you are running the latest admin webapp. Simply shift reload the page before trying to export the config file.

Keep us posted, thanks

Brubeck48, 2022-01-17:
I updated to 6.1.3 and started from terminal but have the same result. Nothing happening when I pushing the button. No errors either in terminal and admins page Logs. No notifications from browser. The same in Chrome and Firefox.

https://user-images.githubusercontent.com/94329654/149822871-8c476615-bae8-4419-a6d4-caf90554f691.mov

chuot, 2022-01-17:
two things:

1. Press F12 and look in the javascript console for any error message that could give us a hint.
2. If you use any sort of a ad blocker extension, can you try disabling it and retry the export?

Thanks

Brubeck48, 2022-01-17:
I have this error in console:

`ERROR Error: Uncaught (in promise): InvalidCharacterError: Failed to execute 'btoa' on 'Window': The string to be encoded contains characters outside of the Latin1 range.
Error: Failed to execute 'btoa' on 'Window': The string to be encoded contains characters outside of the Latin1 range.
    at 842.c52e89dacd2d3418.js:1:122239
    at Generator.next (<anonymous>)
    at s (main.82aacee36c62ecd2.js:1:1016117)
    at fe (main.82aacee36c62ecd2.js:1:1016334)
    at T.invoke (polyfills.8a313204615412e3.js:1:6557)
    at Object.onInvoke (main.82aacee36c62ecd2.js:1:428808)
    at T.invoke (polyfills.8a313204615412e3.js:1:6497)
    at I.run (polyfills.8a313204615412e3.js:1:1953)
    at polyfills.8a313204615412e3.js:1:16750
    at T.invokeTask (polyfills.8a313204615412e3.js:1:7175)
    at G (polyfills.8a313204615412e3.js:1:15940)
    at polyfills.8a313204615412e3.js:1:15024
    at s (main.82aacee36c62ecd2.js:1:1016156)
    at fe (main.82aacee36c62ecd2.js:1:1016334)
    at T.invoke (polyfills.8a313204615412e3.js:1:6557)
    at Object.onInvoke (main.82aacee36c62ecd2.js:1:428808)
    at T.invoke (polyfills.8a313204615412e3.js:1:6497)
    at I.run (polyfills.8a313204615412e3.js:1:1953)
    at polyfills.8a313204615412e3.js:1:16750
    at T.invokeTask (polyfills.8a313204615412e3.js:1:7175)
Vg @ main.82aacee36c62ecd2.js:1
handleError @ main.82aacee36c62ecd2.js:1
next @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
_next @ main.82aacee36c62ecd2.js:1
next @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
next @ main.82aacee36c62ecd2.js:1
emit @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
invoke @ polyfills.8a313204615412e3.js:1
run @ polyfills.8a313204615412e3.js:1
runOutsideAngular @ main.82aacee36c62ecd2.js:1
onHandleError @ main.82aacee36c62ecd2.js:1
handleError @ polyfills.8a313204615412e3.js:1
runGuarded @ polyfills.8a313204615412e3.js:1
i.microtaskDrainDone @ polyfills.8a313204615412e3.js:1
L @ polyfills.8a313204615412e3.js:1
invokeTask @ polyfills.8a313204615412e3.js:1
S @ polyfills.8a313204615412e3.js:1
D @ polyfills.8a313204615412e3.js:1
load (async)
B.p @ polyfills.8a313204615412e3.js:1
scheduleTask @ polyfills.8a313204615412e3.js:1
onScheduleTask @ polyfills.8a313204615412e3.js:1
scheduleTask @ polyfills.8a313204615412e3.js:1
scheduleTask @ polyfills.8a313204615412e3.js:1
scheduleEventTask @ polyfills.8a313204615412e3.js:1
(anonymous) @ polyfills.8a313204615412e3.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
_trySubscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
subscribe @ main.82aacee36c62ecd2.js:1
ee @ main.82aacee36c62ecd2.js:1
me @ main.82aacee36c62ecd2.js:1
_next @ main.82aacee36c62ecd2.js:1
next @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
_trySubscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
subscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
subscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
subscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
g @ main.82aacee36c62ecd2.js:1
subscribe @ main.82aacee36c62ecd2.js:1
(anonymous) @ 842.c52e89dacd2d3418.js:1
t @ polyfills.8a313204615412e3.js:1
Y @ 842.c52e89dacd2d3418.js:1
(anonymous) @ 842.c52e89dacd2d3418.js:1
s @ main.82aacee36c62ecd2.js:1
fe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
t @ polyfills.8a313204615412e3.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
getConfig @ 842.c52e89dacd2d3418.js:1
(anonymous) @ 842.c52e89dacd2d3418.js:1
s @ main.82aacee36c62ecd2.js:1
fe @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
t @ polyfills.8a313204615412e3.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
export @ 842.c52e89dacd2d3418.js:1
(anonymous) @ 842.c52e89dacd2d3418.js:1
Lh @ main.82aacee36c62ecd2.js:1
a @ main.82aacee36c62ecd2.js:1
(anonymous) @ main.82aacee36c62ecd2.js:1
invokeTask @ polyfills.8a313204615412e3.js:1
onInvokeTask @ main.82aacee36c62ecd2.js:1
invokeTask @ polyfills.8a313204615412e3.js:1
runTask @ polyfills.8a313204615412e3.js:1
invokeTask @ polyfills.8a313204615412e3.js:1
S @ polyfills.8a313204615412e3.js:1
D @ polyfills.8a313204615412e3.js:1
Show 57 more frames`

atlex, 2022-01-18:
Hi @chuot i had the same issue some time ago. I know how to reproduce it.
Just try to use not Latin symbols in System/TG/Tags names. It's ABSOLUTELY impossible to export your configuration with e.g. cyrillic symbols. 

@Brubeck48 Am i right?

chuot, 2022-01-18:
I was able to reproduce. Will be working on a fix for v6.1.4.

Brubeck48, 2022-01-30:
Thank you Chrystian. Now export is working

closed 2022-01-30 by Brubeck48