chuot/rdio-scanner #96: 6.1.3 Linux stops ingesting

cfc62, 2022-01-16:
### Discussed in https://github.com/chuot/rdio-scanner/discussions/95

<div type='discussions-op-text'>

<sup>Originally posted by **cfc62** January 16, 2022</sup>
Not a lot of info here, sorry about that. I started another thread on password issues, I decided to give a new Linux box a whirl, and it was great for about 30 minutes, then it stopped ingesting.

There was no error thrown that I could see, and ingesting stopped roughly 10 minutes before there was a "pruning database" message in the logs. Recordings continued to build in the recordings directory for another 10 minutes before I killed the rio-scanner process. Restarting was successful (or has been so far).

No other messages seen. The rdio-scanner process was at 0% cpu when I killed it. 

SDRTrunk also on this same box
Linux (Ubuntu 20.04), all SSD storage, 6 cores 12 threads, 32G RAM
Rdio-scanner 6.1.3
</div>

chuot, 2022-01-17:
Also mentioned [here](https://github.com/chuot/rdio-scanner/discussions/81), I will add a check to the dirwatch thread to make sure it hasn't crashed. If it has, then I will restart a new one. This should fix the issue while the [fsnotify](https://github.com/fsnotify/fsnotify) package gets this resolved.

chuot, 2022-01-31:
Did the latest version resolved your issue?

noamlivne, 2022-02-07:
Happens very rarely in 6.1.7 with Tailscale + Windows10 + SDRTRUNK, when swiping Tailscale and the Android app from Android's memory. Only pruning continues:
2022/02/06 21:30:59 client.conn.writemessage: write tcp 100.79.29.86:443->100.89.248.62:45516: wsasend: An existing connection was forcibly closed by the remote host.

Closing the Windows10 Rdio-scanner program and starting it again helps, and the ingestion cathes up with all the skipped calls.
I'll install the 6.1.10 and keep an eye on it.

closed 2022-02-14 by chuot