chuot/rdio-scanner #99: Service can't read new files in monitored folders

Brubeck48, 2022-01-17:
Hi!

While upgrading to 6.1.3 from 6.1.2 I stopped service rdio-scanner by using kill -9 <id>, sudo ./rdio-scanner -service stop, kill <id>.
But if I want again run the service it's started but I see my monitored folder full of files (deleteAfter on). No sound. No info in Logs about reading new mp3 files. Just only server started and number of listeners...

I tried restart my computer, restart service by sudo ./rdio-scanner -service restart, uninstall service by sudo ./rdio-scanner -service uninstall, install it and start it again but rdio-scanner doesn't work as expected.

If I kill the process and starting from terminal everything working.

And some info for thinking while I updating I lost my yesterdays changes on DB about new TalkGroups and new System. It is not a problem for me but maybe this information will help you to improve your app.

Thank you.

chuot, 2022-01-18:
Don't kill -9 Rdio Scanner when it runs as a service, simply `sudo ./rdio-scanner -service stop`, or `sudo systemctl stop rdio-scanner.service`. This will spare the database in regards to corruption.

It looks like permissions issue. Is there any strange error message if you do `sudo journalctl -u rdio-scanner.service` ?

Brubeck48, 2022-01-20:
Sorry I didn't specify: this problem in macOS Monterey.

I tried to find some strange errors while I'm starting service by mac's Console app but unfortunately (or fortunately 🙂) no errors in logs.

chuot, 2022-01-31:
Did the latest version resolved your issue?

Brubeck48, 2022-01-31:
Unfortunately no. The problem still exist on v6.1.8.

Brubeck48, 2022-01-31:
Some new systems logs from last starting:

```
Heaviest stack for the target process:
  3  thread_start + 15 (libsystem_pthread.dylib + 8207) [0x7ff80a78e00f]
  3  _pthread_start + 125 (libsystem_pthread.dylib + 25844) [0x7ff80a7924f4]
  3  ??? [0x1061725]
  3  ??? [0x103868c]
  3  ??? [0x1038753]
  2  ??? [0x1040dbb]
  2  ??? [0x100b68a]
  2  ??? [0x100b4a9]
  2  ??? [0x102f770]
  2  ??? [0x1051d34]
  2  ??? [0x106374d]
  2  ??? [0x10656f4]
  2  __psynch_cvwait + 10 (libsystem_kernel.dylib + 13574) [0x7ff80a758506]
```
and:
```
 5   <missing stack>
    5   <Requested Thread QoS Utility>
  5   <missing stack>
  3   thread_start + 15 (libsystem_pthread.dylib + 8207) [0x7ff80a78e00f]
    3   _pthread_start + 125 (libsystem_pthread.dylib + 25844) [0x7ff80a7924f4]
      3   ??? [0x1061725]
        3   ??? [0x103868c]
          3   ??? [0x1038753]
            2   ??? [0x1040dbb]
              2   ??? [0x100b68a]
                2   ??? [0x100b4a9]
                  2   ??? [0x102f770]
                    2   ??? [0x1051d34]
                      2   ??? [0x106374d]
                        2   ??? [0x10656f4]
                          2   __psynch_cvwait + 10 (libsystem_kernel.dylib + 13574) [0x7ff80a758506]
                            2   <Kernel mode>
            1   ??? [0x1040ca5]
              1   ??? [0x1039a58]
  1   <missing stack>
    1   <Kernel mode, Requested Thread QoS Utility>

  Binary Images:
             0x4ad8000 -          0x4b5bfff  dyld (941.4)                       <CEF5A27A-D50B-3020-AF03-1734B19BC8C5>  /usr/lib/dyld
        0x7ff80a755000 -     0x7ff80a78bfff  libsystem_kernel.dylib (8019.61.5) <5AA1E5BE-B5B8-3A02-9885-A8C99E0CA378>  /usr/lib/system/libsystem_kernel.dylib
        0x7ff80a78c000 -     0x7ff80a797fff  libsystem_pthread.dylib (485.60.2) <6C7561B4-4B92-3F45-921E-ABE669299844>  /usr/lib/system/libsystem_pthread.dylib
```

chuot, 2022-01-31:
I never asked which software you're using to generate the files you are trying to ingest? I tried to install SDRTrunk on my M1 Mac but did not succeed. So I tried Rdio Scanner v6.1.8 as a service on my M1 Mac with Monterey 12.2. While monitoring a folder, I dumped a MP3 file generated by SDRTrunk from another machine to that folder and it was ingested no problem.

Could you look in the logs from the Rdio Scanner Admin Dashboard if there's is something about failing to ingest files from your monitored folder? I've added some outputs to the logs in regards to that sinve v6.1.8.

Also make sure that you are really running the latest v6.1.8 because the ingestion problem that started with v6.1.3 has been resolved with v6.1.8. It was a stupid typo error on the server side. When you unzip the newer executable over the old one, be sure that the service has been stopped before the unzip, then restart the service.

chuot, 2022-02-21:
No updates. Closing

closed 2022-02-21 by chuot