chuot/rdio-scanner #100: logfile for process death

cfc62, 2022-01-18:
Just died with a fatal error, log below:

fatal error: concurrent map read and map write

goroutine 37 [running]:
runtime.throw({0xa64ba8, 0xc00043e750})
	/home/chuot/go/src/runtime/panic.go:1198 +0x71 fp=0xc0002d3ce0 sp=0xc0002d3cb0 pc=0x435151
runtime.mapaccess1_fast64(0xc000131d48, 0xc00019dd40, 0x34)
	/home/chuot/go/src/runtime/map_fast64.go:21 +0x172 fp=0xc0002d3d00 sp=0xc0002d3ce0 pc=0x412a12
main.(*LivefeedMap).IsEnabled(0xc0002320d8, 0xc00009f320)
	/home/chuot/Dev/rdio-scanner/server/livefeed.go:73 +0x67 fp=0xc0002d3d48 sp=0xc0002d3d00 pc=0x97dfa7
main.(*Controller).EmitCall(0xc0001ea000, 0xc00009f320)
	/home/chuot/Dev/rdio-scanner/server/controller.go:152 +0xe5 fp=0xc0002d3de0 sp=0xc0002d3d48 pc=0x96cf65
main.(*Controller).IngestCall(0xc0001ea000, 0xc00009f320)
	/home/chuot/Dev/rdio-scanner/server/controller.go:362 +0xec6 fp=0xc0002d3f38 sp=0xc0002d3de0 pc=0x96e026
main.(*Controller).Start.func2()
	/home/chuot/Dev/rdio-scanner/server/controller.go:620 +0xff fp=0xc0002d3fe0 sp=0xc0002d3f38 pc=0x96fc5f
runtime.goexit()
	/home/chuot/go/src/runtime/asm_amd64.s:1581 +0x1 fp=0xc0002d3fe8 sp=0xc0002d3fe0 pc=0x4651a1
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:616 +0x2bb

goroutine 1 [IO wait, 67 minutes]:
internal/poll.runtime_pollWait(0x7f373d51d6d8, 0x72)
	/home/chuot/go/src/runtime/netpoll.go:234 +0x89
internal/poll.(*pollDesc).wait(0xc0002b9580, 0x4186e6, 0x0)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:84 +0x32
internal/poll.(*pollDesc).waitRead(...)
	/home/chuot/go/src/internal/poll/fd_poll_runtime.go:89
internal/poll.(*FD).Accept(0xc0002b9580)
	/home/chuot/go/src/internal/poll/fd_unix.go:402 +0x22c
net.(*netFD).accept(0xc0002b9580)
	/home/chuot/go/src/net/fd_unix.go:173 +0x35
net.(*TCPListener).accept(0xc00028a210)
	/home/chuot/go/src/net/tcpsock_posix.go:140 +0x28
net.(*TCPListener).Accept(0xc00028a210)
	/home/chuot/go/src/net/tcpsock.go:262 +0x3d
net/http.(*Server).Serve(0xc0002d8000, {0xccf738, 0xc00028a210})
	/home/chuot/go/src/net/http/server.go:3002 +0x394
net/http.(*Server).ListenAndServe(0xc0002d8000)
	/home/chuot/go/src/net/http/server.go:2931 +0x7d
net/http.ListenAndServe(...)
	/home/chuot/go/src/net/http/server.go:3185
main.main()
	/home/chuot/Dev/rdio-scanner/server/main.go:230 +0x9e5

goroutine 19 [select, 95 minutes]:
database/sql.(*DB).connectionOpener(0xc00011de10, {0xcd1658, 0xc0001c0c40})
	/home/chuot/go/src/database/sql/sql.go:1196 +0x93
created by database/sql.OpenDB
	/home/chuot/go/src/database/sql/sql.go:794 +0x188

goroutine 34 [select, 95 minutes]:
main.(*Admin).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/admin.go:568 +0xe7
created by main.(*Admin).Start
	/home/chuot/Dev/rdio-scanner/server/admin.go:566 +0x69

goroutine 35 [select, 35 minutes]:
main.(*Scheduler).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:63 +0x77
created by main.(*Scheduler).Start
	/home/chuot/Dev/rdio-scanner/server/scheduler.go:61 +0xa7

goroutine 36 [chan receive, 95 minutes]:
main.(*Controller).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/controller.go:612 +0x71
created by main.(*Controller).Start
	/home/chuot/Dev/rdio-scanner/server/controller.go:609 +0x275

goroutine 38 [syscall]:
syscall.Syscall6(0xe8, 0x9, 0xc00030fbec, 0x7, 0xffffffffffffffff, 0x0, 0x0)
	/home/chuot/go/src/syscall/asm_linux_amd64.s:43 +0x5
golang.org/x/sys/unix.EpollWait(0x2, {0xc00030fbec, 0x2, 0xc0001965f0}, 0xc00030fcb4)
	/home/chuot/go/pkg/mod/golang.org/x/sys@v0.0.0-20220111092808-5a964db01320/unix/zsyscall_linux_amd64.go:56 +0x58
github.com/fsnotify/fsnotify.(*fdPoller).wait(0xc0002a0060)
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.1/inotify_poller.go:87 +0x7d
github.com/fsnotify/fsnotify.(*Watcher).readEvents(0xc00028c4b0)
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.1/inotify.go:193 +0x2b0
created by github.com/fsnotify/fsnotify.NewWatcher
	/home/chuot/go/pkg/mod/github.com/fsnotify/fsnotify@v1.5.1/inotify.go:60 +0x1c7

goroutine 39 [chan receive]:
main.(*Dirwatch).Start.func1()
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:472 +0xf6
created by main.(*Dirwatch).Start
	/home/chuot/Dev/rdio-scanner/server/dirwatch.go:458 +0x125

goroutine 21 [syscall, 95 minutes]:
os/signal.signal_recv()
	/home/chuot/go/src/runtime/sigqueue.go:169 +0x98
os/signal.loop()
	/home/chuot/go/src/os/signal/signal_unix.go:24 +0x19
created by os/signal.Notify.func1.1
	/home/chuot/go/src/os/signal/signal.go:151 +0x2c

goroutine 6879 [select]:
main.(*Client).Init.func2()
	/home/chuot/Dev/rdio-scanner/server/client.go:91 +0x128
created by main.(*Client).Init
	/home/chuot/Dev/rdio-scanner/server/client.go:82 +0x1be

goroutine 6878 [runnable]:
main.(*Controller).ProcessMessageCommandLivefeedMap(0x6e5ec6, 0xc000232080, 0xc00028a108)
	/home/chuot/Dev/rdio-scanner/server/controller.go:455 +0xbd
main.(*Controller).ProcessMessage(0xc000387fa0, 0xc000232080, 0x390)
	/home/chuot/Dev/rdio-scanner/server/controller.go:398 +0x1f8
main.(*Client).Init.func1()
	/home/chuot/Dev/rdio-scanner/server/client.go:73 +0x108
created by main.(*Client).Init
	/home/chuot/Dev/rdio-scanner/server/client.go:58 +0x165

...and then dropped back to the shell prompt.

cfc62, 2022-01-18:
Tried to input the errors as code as github is losing the initial indenting, not working for some reason, sorry

chuot, 2022-01-18:
Was this with v6.1.3?

You can get the version number with: `./rdio-scanner -version`.

cfc62, 2022-01-18:
Yes, 6.1.3

chuot, 2022-01-20:
I think it's a concurency issue. I have had some mutex on the upcoming version. Hopefully this will resolve your issue.

cfc62, 2022-01-21:
I will say I suspect this happened when probably 20 or even 30 files arrived all at the same time (noise burst).

chuot, 2022-01-31:
Should be fixed in the latest version, can you confirm?

closed 2022-02-14 by chuot