chuot/rdio-scanner #103: Frequency Does not update in IOS app but is correct on web gui

tdduarte, 2022-01-19:
When adding fake frequency into rdio admin page (had to use for voxcall as the frequency entered doesn't seem to push to rdio) it updates the web gui but does not update on the ios app. 
[image]

tdduarte, 2022-01-19:
[image]

chuot, 2022-01-19:
Please confirm that you've entered the frequency at the dirwatch level, or was it in the talkgroup definition?

tdduarte, 2022-01-19:
This is in the talkgroup definition as its an api from voxcall

chuot, 2022-01-19:
Thanks, it will be fixed in the next release of the native app.

tdduarte, 2022-01-19:
Great!  Thank you for all your work on this project!

closed 2022-01-31 by chuot