chuot/rdio-scanner #104: Can't upload to 6.x

tadscottsmith, 2022-01-19:
I'm running two instances of rdio-scanner in docker side-by-side. One of them is the original 5.2.9 version, and the other is a version of 6.1.3 that was upgraded from the 5.2.9 version. In trunk-recorder, I'm using my upload script to call the /api/trunk-recorder-call-upload on each version. However, no matter what I've tried, nothing seems to be getting imported into the 6.1.3 version. Curl indicates the files are uploading successfully to both, but only 5.2.9 actually logs that a new call has been imported. Is there any additional logging I can enable on 6.1.3 to see what's being missed? I'm able to see and play the historical recordings in 6.1.3 that were present when I upgraded it, but no new recordings ever appear.

tadscottsmith, 2022-01-19:
Actually, it looks like maybe the session is never being closed. It just hangs here.

curl -v -k https://localhost:3001/api/trunk-recorder-call-upload \
     -F "key=<REDACTED>" \
     -F "audio=906-1642633779_853212500-call_8430.wav;type=audio/wav" \
     -F "906-1642633779_853212500-call_8430.json;type=application/json" \
     -F "system=1"
   Trying 127.0.0.1:3001...
 TCP_NODELAY set
 Connected to localhost (127.0.0.1) port 3001 (#0)
 ALPN, offering h2
 ALPN, offering http/1.1
 successfully set certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
 TLSv1.3 (OUT), TLS handshake, Client hello (1):
 TLSv1.3 (IN), TLS handshake, Server hello (2):
 TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
 TLSv1.3 (IN), TLS handshake, Certificate (11):
 TLSv1.3 (IN), TLS handshake, CERT verify (15):
 TLSv1.3 (IN), TLS handshake, Finished (20):
 TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
 TLSv1.3 (OUT), TLS handshake, Finished (20):
 SSL connection using TLSv1.3 / TLS_AES_128_GCM_SHA256
 ALPN, server accepted to use h2
 Server certificate:
  subject: CN=<REDACTED>
  start date: Dec 23 17:36:10 2021 GMT
  expire date: Mar 23 17:36:09 2022 GMT
  issuer: C=US; O=Let's Encrypt; CN=R3
  SSL certificate verify result: unable to get local issuer certificate (20), continuing anyway.
 Using HTTP2, server supports multi-use
 Connection state changed (HTTP/2 confirmed)
 Copying HTTP/2 data in stream buffer to connection buffer after upgrade: len=0
 Using Stream ID: 1 (easy handle 0x55bf4e2acd50)
 POST /api/trunk-recorder-call-upload HTTP/2
 Host: localhost:3001
 user-agent: curl/7.68.0
 accept: */*
 content-length: 576
 content-type: multipart/form-data; boundary=------------------------e4d149400df0b231

 TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
 Connection state changed (MAX_CONCURRENT_STREAMS == 250)!
 We are completely uploaded and fine

chuot, 2022-01-20:
I checked the code and you should have had a response back, unless something is crashing on the server side.

Can you get the docker container console outputs, they may have a hint on what's going on.

tadscottsmith, 2022-01-24:
Nothing is ever logged on the console (no |info|NewCall:) It's like almost like the upload is completing, but rdio-scanner is never acknowledging the completed upload and closing the session. Inside the container I can see all of the sessions established, but only one of them has received data. Is there any additional logging that can be enabled? Or do I need to build from scratch to enable some logging in the API? 

`/app/data # netstat
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41126 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41026 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:40800 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:40916 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41342 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41186 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:40772 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41334 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41218 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41162 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41346 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:40764 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41194 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41012 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:40776 ESTABLISHED
tcp   134609      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41350 ESTABLISHED
tcp        0      0 f2deed1d70c8:3000       trunk-recorder.radio_default:41182 ESTABLISHED`

chuot, 2022-01-24:
I tested the docker container image v6.1.4 to make sure and it works without issue.

For your information, I was running it from the command line as follows:

```
docker run --detach --publish 3000:3000 --volume ~/.rdio-scanner:/app/data:z --name rdio-scanner docker.io/chuot/rdio-scanner:latest
```

Then I looked at your upload script again, to realize that you were missing the `@ sign` in front of the path to the wav and json files. Note that the `meta=` field is also missing. You should have this instead:

```
-F "audio=@906-1642633779_853212500-call_8430.wav;type=audio/wav"
-F "meta=@906-1642633779_853212500-call_8430.json;type=application/json"
```

I guess that explains why it worked for you. I'll make sure this will be reported as an error in the next release.

I have also bringed back the `docs/examples` folder which contains an upload-script.sh example.

taclane, 2022-01-24:
I'm unable to say what exactly fixed it, but I was noticing the exact same issue until the latest 6.1.6 release.

I had a docker container running 6.1.4 and it had all the above problems, and was consuming > 85% of the host CPU trying to resolve uploads pointed at the `trunk-recorder-call-upload` endpoint.

tadscottsmith, 2022-01-24:
I can confirm the original script is now working with 6.1.6. I was seeing the same symptoms as taclane, as well as the calls not being imported.

I did miss the @ sign in my manual curl commands above, but the @ sign was present in my upload script and it still was not working prior to 6.1.6.

closed 2022-01-26 by tadscottsmith