chuot/rdio-scanner #115: Talk Group Names not Shown

tadscottsmith, 2022-01-26:
It appears the talk group 'label' is being shown twice, and the 'name' is not being shown.

[image]

chuot, 2022-01-26:
what's in the admin interface in regards to those talkgroups? What do they have in the name field?

If they where autopopulated from trunk-recorder, then it's normal because I only have the talkgroup_tag value, which I'm using for both label and name.

tadscottsmith, 2022-01-26:
Actually, I just checked the DB and they're both the same even though they had separate fields in the CSV. This may be related to the import process, so let me do some digging.

tadscottsmith, 2022-01-26:
So, exactly what you mentioned is the problem. Instead of "autopopulating" any talk groups that do not exist, it appears to be overwriting existing talk groups with the value from talkgroup_tag as both label and name. In my opinion, the autopopulate feature shouldn't overwrite an existing entry.

chuot, 2022-01-27:
Oh now I see. This new feature to modify on the fly the talkgroup label and name was recently added in v6.1.4 as a convinience feature for SDRTrunk users. This seemed useful since users can modify labels directly under SDRTrunk. As for Trunk Recorder, it's a nuisance more than other things since Trunk Recorder doesn't have a configuration interface like SDRTrunk, which forces Trunk Recorder users to keep config files up to date as well as the talkgroups csv files so that the information on the Rdio Scanner side is compliant. In short, what you are currently experiencing. Know that I rely a lot on the autopopulate function to make life easier for users, regardless of whether they come from the Trunk Recorder or SDRTrunk world. Since v6.1.7 is already well underway, I'll be sure to review how things are done with respect to labels and talkgroup names. I don't know yet if it will be in the form of a global option or if this function of automatic updating of labels and names will only be for SDRTrunk. Hoping that everyone will agree with the decision that I will take.

tadscottsmith, 2022-01-27:
I think the confusion for me was mostly in the naming. To me "Auto-populate" would be to bring in talk groups that don't exist, but wouldn't update existing groups. Something like "Auto-update" would make more sense to make changes to existing talk groups. I see you made some changes to this in 6.1.7, so I'll give it a shot.

tadscottsmith, 2022-02-14:
Yeah, I will go ahead and close. Thank you!

closed 2022-02-14 by tadscottsmith