chuot/rdio-scanner #119: Sort Problem

Dewey3, 2022-01-29:
Am I doing something wrong?  I have a preferred sort order that is not TGID order, so I keep it in a CSV file.  I can import that TG CSV file and when I look at the order of the talkgroups in the Admin System section, my sort is preserved.  However, once I'm running the scanner, if I click on the Select TG button, the entries are sorted by TGID.  At that point, if I go into the Admin panel and view the talkgroups in the Systems section, they are also now sorted by TGID.  If this makes a difference, I am running 3 RPis.  2 of the RPis are running Trunk-Recorder only and send the recordings to the 3rd RPi that only runs Rdio-Scanner.  It is that 3rd RPi that I cannot seem to make my preferred sort order stick.

chuot, 2022-01-31:
Can you confirm the `Config / Options / Sort Talkgroups` is deactivated?

Dewey3, 2022-01-31:
Yes sir, Options > Sort Talkgroups is turned off.  I don't know if it makes a difference, but Systems > Auto Populate is turned on.   The only reason I bring this up is because the CSV file does import in my preferred sort, but somehow changes on its own after everything is running.  My limited thinking is that maybe it's related to auto populate, but I just don't know enough to say.  What I will do later today since I am running two different systems into this is to disable auto populate on one system while keeping it enabled on the other system.  Just trying to do my own troubleshooting in case it helps.

chuot, 2022-01-31:
Thanks for the update. I will review the code in regards to the autopopulate option. Something could be wrong.

Keep us posted if you find anything else.

Dewey3, 2022-01-31:
Ok, maybe it's not auto populate.  My steps and results:

- Stopped both RPis running Trunk-Recorder so they don't keep trying to feed Rdio-Scanner while I troubleshoot
- Went in Rdio-Scanner's Admin console and deleted both systems (ID 1 & ID 2), saved and restarted the admin panel
- Loaded the CSVs for the first system, looked at import page and my preferred sort was retained, so I competed the import
- Went into the "New System" of the imported CSV file and configured it using the same IDs as the previously deleted systems
- Ensured Auto Populate was turned off
- Did all of the same for the second system

Once finished and looking at the talkgroups in the System panel, they are again sorted by TGID instead of how they were displayed on the import page.

chuot, 2022-01-31:
I found the issue. It will be fixed in the upcoming v6.1.9.

Dewey3, 2022-02-01:
I can confirm that this is now resolved in v6.1.9.  Thank you.

closed 2022-02-01 by Dewey3