chuot/rdio-scanner #120: v6.1.8

JayManCanada6014, 2022-02-01:
Brand no 6.1.8, no importing of any old settings. The overall system auto-populate seems to be on by default. I turned it off and deleted the two systems it made and Rdio-Scanner crashed.

Jan 31 19:42:47 vmi767181 rdio-scanner[2440653]: 2022/01/31 19:42:47 newcall: system=201 talkgroup=2039 file=2039-1643676158_867637500-call_2689.wav no matching system/talkgroup
Jan 31 19:42:48 vmi767181 rdio-scanner[2440653]: 2022/01/31 19:42:48 newcall: system=201 talkgroup=1506 file=1506-1643676142_867537500-call_2686.m4a success
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: panic: runtime error: invalid memory address or nil pointer dereference
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x975b54]
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: goroutine 31 [running]:
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: main.(*Controller).IngestCall(0xc0000b6f00, 0xc0003b8fc0)
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: #011/home/chuot/Dev/rdio-scanner/server/controller.go:214 +0x394
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: main.(*Controller).Start.func2()
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: #011/home/chuot/Dev/rdio-scanner/server/controller.go:641 +0x105
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: created by main.(*Controller).Start
Jan 31 19:42:52 vmi767181 rdio-scanner[2440653]: #011/home/chuot/Dev/rdio-scanner/server/controller.go:637 +0x2d1
Jan 31 19:42:52 vmi767181 systemd[1]: rdio-scanner.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jan 31 19:42:52 vmi767181 systemd[1]: rdio-scanner.service: Failed with result 'exit-code'.

JayManCanada6014, 2022-02-01:
And then it restarted itself as the service should and it crashed again after a bit.

Jan 31 19:42:52 vmi767181 systemd[1]: rdio-scanner.service: Failed with result 'exit-code'.
Jan 31 19:44:52 vmi767181 systemd[1]: rdio-scanner.service: Scheduled restart job, restart counter is at 2.
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: Rdio Scanner v6.1.8
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: ----------------------------------
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:52 server started
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:52 base folder is /opt/rdio-scanner
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:52 main interface at http://vmi767181:3000
Jan 31 19:44:52 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:52 admin interface at http://vmi767181:3000/admin
Jan 31 19:44:53 vmi767181 tailscaled[1421791]: control: HostInfo: {"IPNVersion":"1.20.2-t312750ddd-g2c164d9c7","BackendLogID":"5190f40c7c1f4cc5a1b14ca5569df2a6280bf22e9b76d1503d7c809fde47831a","OS":"linux","OSVersion":"Ubuntu 20.04.3 LTS (Focal Fossa); kernel=5.4.0-92-generic","Hostname":"vmi767181","GoArch":"amd64","Services":[{"Proto":"tcp","Port":22,"Description":"sshd -D [listener] 0 of 10-100 startups"},{"Proto":"tcp","Port":53,"Description":"systemd-resolved"},{"Proto":"tcp","Port":3000,"Description":"rdio-scanner"},{"Proto":"tcp","Port":46297,"Description":"tailscaled"},{"Proto":"peerapi4","Port":46297},{"Proto":"peerapi6","Port":46297},{"Proto":"peerapi-dns-proxy","Port":1}],"NetInfo":{"MappingVariesByDestIP":false,"HairPinning":false,"WorkingIPv6":true,"WorkingUDP":true,"UPnP":false,"PMP":false,"PCP":false,"PreferredDERP":1,"DERPLatency":{"1-v4":0.003890308,"1-v6":0.003354163,"10-v4":0.067275088,"10-v6":0.067222589,"11-v4":0.114438954,"11-v6":0.11694863,"12-v4":0.020782125,"12-v6":0.020237055,"2-v4":0.071037678,"2-v6":0.072289,"4-v4":0.097256048,"4-v6":0.08711756,"8-v4":0.07311935,"9-v4":0.040946264,"9-v6":0.041250369}}}
Jan 31 19:44:53 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:53 newcall: system=100 talkgroup=8224 file=8224-1643676282_142125000-call_295.m4a success
Jan 31 19:44:54 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:54 listeners count is 1
Jan 31 19:44:57 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:57 listeners count is 2
Jan 31 19:44:59 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:44:59 newcall: system=201 talkgroup=1506 file=1506-1643676282_867537500-call_2709.m4a success
Jan 31 19:45:12 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:12 newcall: system=102 talkgroup=35472 file=35472-1643676304_141525000-call_1825.wav no matching system/talkgroup
Jan 31 19:45:12 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:12 newcall: system=102 talkgroup=35440 file=35440-1643676304_141375000-call_1826.wav no matching system/talkgroup
Jan 31 19:45:15 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:15 newcall: system=102 talkgroup=35440 file=35440-1643676307_142425000-call_1827.wav no matching system/talkgroup
Jan 31 19:45:17 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:17 newcall: system=201 talkgroup=1506 file=1506-1643676298_867637500-call_2714.m4a success
Jan 31 19:45:18 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:18 newcall: system=102 talkgroup=35440 file=35440-1643676310_142005000-call_1828.wav no matching system/talkgroup
Jan 31 19:45:19 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:19 newcall: system=102 talkgroup=35456 file=35456-1643676313_141735000-call_1829.wav no matching system/talkgroup
Jan 31 19:45:20 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:20 newcall: system=102 talkgroup=35440 file=35440-1643676313_142995000-call_1830.wav no matching system/talkgroup
Jan 31 19:45:27 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:27 newcall: system=102 talkgroup=35456 file=35456-1643676318_141375000-call_1831.wav no matching system/talkgroup
Jan 31 19:45:28 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:28 newcall: system=201 talkgroup=1506 file=1506-1643676314_867537500-call_2715.m4a success
Jan 31 19:45:30 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:30 newcall: system=102 talkgroup=35440 file=35440-1643676321_141525000-call_1832.wav no matching system/talkgroup
Jan 31 19:45:33 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:33 newcall: system=102 talkgroup=35440 file=35440-1643676325_141735000-call_1835.wav no matching system/talkgroup
Jan 31 19:45:34 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:34 newcall: system=100 talkgroup=8896 file=8896-1643676325_142425000-call_1833.m4a success
Jan 31 19:45:35 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:35 newcall: system=100 talkgroup=8896 file=8896-1643676325_141435000-call_296.wav duplicate call rejected
Jan 31 19:45:36 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:36 newcall: system=102 talkgroup=35680 file=35680-1643676325_142950000-call_297.wav no matching system/talkgroup
Jan 31 19:45:38 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:38 newcall: system=201 talkgroup=1502 file=1502-1643676324_867387500-call_2718.wav no matching system/talkgroup
Jan 31 19:45:40 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:40 newcall: system=102 talkgroup=35680 file=35680-1643676330_141675000-call_298.wav no matching system/talkgroup
Jan 31 19:45:43 vmi767181 rdio-scanner[2440993]: 2022/01/31 19:45:43 newcall: system=102 talkgroup=35456 file=35456-1643676331_141525000-call_1838.wav no matching system/talkgroup
[... cut ...]
Jan 31 19:47:42 vmi767181 rdio-scanner[2440993]: #011/home/chuot/Dev/rdio-scanner/server/controller.go:641 +0x105
Jan 31 19:47:42 vmi767181 rdio-scanner[2440993]: created by main.(*Controller).Start
Jan 31 19:47:42 vmi767181 rdio-scanner[2440993]: #011/home/chuot/Dev/rdio-scanner/server/controller.go:637 +0x2d1
Jan 31 19:47:42 vmi767181 systemd[1]: rdio-scanner.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Jan 31 19:47:42 vmi767181 systemd[1]: rdio-scanner.service: Failed with result 'exit-code'.

chuot, 2022-02-01:
Fixed in the upcoming v6.1.9.