chuot/rdio-scanner #123: Auto Populate ... FYI

Dewey3, 2022-02-08:
This is certainly no problem, but it is something I wanted to relay to you as FYI only.  The primary system that I monitor, Prince George's Co, MD, 2A8, P25, Phase 2, TDMA, uses dynamic talkgroups in the 20### numbers that rotate on a daily basis, so I'm always adding new 20### talkgroups to Trunk-Recorder's CSV talkgroup file as priority 255 so they are ignored in the future.  (479 talkgroups in my current 741 talkgroup file are these dynamic talkgroups set to a priority of 255).  Every now and again, a new dynamic talkgroup that I had not already set to ignore gets assigned.  Since I like to keep Auto Populate turned on, which has been very useful, the previously unidentified talkgroup gets ingested by Rdio-Scanner.  Rdio-Scanner properly tags the newly unrecognized talkgroup as "Untagged"; however, it randomly assigns anything, including unprintable characters in the "Groups" field for the new talkgroup.

Dewey3, 2022-02-08:
`Can you post...` this time it was just the single capital letter "V", the unprintable characters will sometimes appear instead, I just can't identify when that will happen (I didn't think you wanted a screenshot of the single V, but can provide it if it helps).  Also if it helps, I can delete some of the 255 priority talkgroups from the CSV file for a day or two to see how other unrecognized talkgroups will be classified.

`Do you run your...` default database engine and database file.

`How do you feed Rdio Scanner...` plugin.

Dewey3, 2022-02-09:
An interesting thing happened... I was posting an issue over on Trunk-Recorder just before posting this, so when I went to answer this, I accidentally placed the answer to this with a link to the screenshots in my Trunk-Recorder thread (https://github.com/robotastic/trunk-recorder/issues/634#issuecomment-1032930131).  Well as a result of that, it looks like this is actually a Trunk-Recorder issue with the Rdio-Scanner plugin anyway.  So my guess is that the rest of this story here is moot.  In case you wanted to see my screenshots of the unprintable characters without having to follow the link over to Trunk-Recorder, the screenshots are here:  https://drive.google.com/drive/folders/1gAUouj7xlJr8RLRx1Ztl6BOhNw4gWfrZ?usp=sharing

taclane, 2022-02-09:
Yeah, here's an example that caused me to investigate this as well.  Previous TG groups slowly got replaced by invisible characters, single letters and the like.
[image]
Root cause seems to be rdio-scanner's plugin for trunk recorder. `tg_group` is a string, but it is getting assigned the TG ID integer without type conversion.  Rdio-scanner wasn't auto-populating the talkgroup group field until somewhat recently, so that's probably why it went unnoticed.

Changing line 62 of the plugin to:
```
      tg_group = std::to_string(call_info.talkgroup);
```
takes care of the junk characters, but it is still just using the TG ID for the group field.  I'm planning to submit a PR so that plugins have all the talkgroup data so that we can auto-populate rdio-scanner completely out of trunk-recorder's talkgroup csv.

Dewey3, 2022-02-09:
This looks like a very east fix for me as an end user.  Thank you for all of your help and support!

closed 2022-02-09 by Dewey3